eblot/pyftdi #220: FT232H: not sure how to access all 16 GPIOs

ntimeu, 2020-10-05:
Hello,

I have been trying to access all 16 GPIOS (ADBUS0-7 and ACBUS0-7), but I'm not sure how to do that.

I managed to access the ADBUS and write values to it using the GpioAsyncController.
But I have issues finding how to read and write values from the ACBUS, I cannot figure out how to get all 16 bits through the GpioAsyncController or GpioMpsseController (or another class).

The documentation says that (https://eblot.github.io/pyftdi/gpio.html)[GPIO] there is only a single, 16-bit wide port, but I cannot find a way to use it to extract all 16 bits.

ntimeu, 2020-10-07:
My bad, I finally found some examples on how to access all GPIOs. Sorry for the time lost :(

closed 2020-10-07 by ntimeu

mpratt14, 2020-10-07:
@ntimeu link what you found here for others

ntimeu, 2020-10-08:
[https://raw.githubusercontent.com/adafruit/Adafruit_Blinka/master/src/adafruit_blinka/microcontroller/ft232h/pin.py](I used this link).

At first I thought that to access GPIOs, I needed to use the pyftdi.gpio API, but on my model, the FT232H, we have 16-bits port, and the GPIO API can only address 8 bits. However, D0, D1 and D2 are reserved for I2C usage and thus cannot be addressed.

eblot, 2020-10-08:
Hi,

First point I should stress on: always specify *which* device you having trouble with, as there is no two FTDI devices with the same capabilities. PyFtdi tries to abstract those differences, but it cannot do better than the HW.

16-bit wide port can only be used through the MPSSE mode, that it either one of the [I2C](https://eblot.github.io/pyftdi/api/i2c.html#pyftdi.i2c.I2cGpioPort) or [SPI](https://eblot.github.io/pyftdi/api/spi.html#pyftdi.spi.SpiGpioPort) controllers, or through the [GpioMpsseController](https://eblot.github.io/pyftdi/api/gpio.html#pyftdi.gpio.GpioMpsseController) controller.

There are alternative modes available on some of the FTDI devices, where the lower 8 bits can be accessed through GPIO "bit bang" APIs, either [asynchronous](https://eblot.github.io/pyftdi/api/gpio.html#pyftdi.gpio.GpioAsyncController) or [synchronous](https://eblot.github.io/pyftdi/api/gpio.html#pyftdi.gpio.GpioSyncController); while the 8 upper bits can controlled through the "[CBUS](https://eblot.github.io/pyftdi/gpio.html#cbus-gpios)" mode on some device. This latter mode does not allow to drive all 16 bits at once, it better suits some kind of low speed GPIO (such as driving a reset or a led) on the upper bits, and only use the lowest 8 bits for "real" communication. A good reference is the matching [data sheet](https://www.ftdichip.com/Support/Documents/DataSheets/ICs/DS_FT232H.pdf) for the FTDI device - but good luck to compare two FTDI devices :-)

D0/D1/D2 are *not* reserved for I2C if you do not need I2C. Use [GpioMpsseController](https://eblot.github.io/pyftdi/api/gpio.html#pyftdi.gpio.GpioMpsseController) if you want to drive all 16 bits on a FT232H.

The answer would have been quite different with a FT232R, hence the need to report the device type :-)

eblot, 2020-10-08:
Note that I'm always open to PR to improve the documentation, helping sorting out FTDI's devices/features spaghetti matrix...

ntimeu, 2020-10-08:
@eblot actually, I need to use D0/1/2 for a I2C chip measuring current, and the others as GPIOs.

BTW, thanks for writing this library, working with FTDI's doc is not really what I would call "easy".