eblot/pyftdi #221: pyftdi segfaults on FreeBSD 12.1

btigner, 2020-10-09:
python 3.7.9 running on FreeBSD
running interactively

>>> from pyftdi.spi import SpiController
>>> spi=SpiController()
>>> spi.configure('ftdi://::/1')
Segmentation fault

I have tried this with 2 ftdi parts
[('FTDI', 'C232HM-EDHSL-0', 'FTU9YGWV')]
and
[('FTDI', 'UMFT220XA', 'FTW9ZN5W')]

I really would like to get this working. Can someone help please ?

eblot, 2020-10-09:
PyFtdi is unlikely to segfault on its own, as it is a pure Python module.
It however relies on PyUSB which in turn uses the native libusb library through `ctypes` module. They have been some unstable `libusb` releases, and some compatibility issues between what PyUSB thinks `libusb` exposes.
Some older version of PyFtdi used to run cleanup calls on exit, which collided with pyusb own cleanup calls - this was a source of potential memory corruption in `libusb`.

Please report:
- which version of PyFTDI are you using
- which version of PyUSB is installed
- which version of libusb is installed on your system

You may also try to run the Python interpreter from a debugger to get useful callstacks, ...

See also #122

eblot, 2020-10-14:
PyUSB ticket: https://github.com/pyusb/pyusb/issues/333

eblot, 2021-01-01:
It seems it is not a PyFTDI issue. Feel free to re-open if it is specific to PyFTDI.

closed 2021-01-01 by eblot