eblot/pyftdi #223: LD_LIBRARY_PATH not adhered on Raspberry Pi

tmplt, 2020-11-09:
I have an Adafruit FT232H breakout I want to use with my Raspberry Pi, an aarch64-linux system. I'm also using NixOS, and must specify `LD_LIBRARY_PATH = "${libusb}/lib/"`.
```
$ echo $LD_LIBRARY_PATH
/nix/store/z1j25bh18iaan59r4s8dn49lak06a582-libusb-1.0.23/lib/
$ ls -l $LD_LIBRARY_PATH
dr-xr-xr-x 2 root root 4.0K Jan  1  1970 .
dr-xr-xr-x 3 root root 4.0K Jan  1  1970 ..
-r-xr-xr-x 1 root root  989 Jan  1  1970 libusb-1.0.la
lrwxrwxrwx 1 root root   19 Jan  1  1970 libusb-1.0.so -> libusb-1.0.so.0.2.0
lrwxrwxrwx 1 root root   19 Jan  1  1970 libusb-1.0.so.0 -> libusb-1.0.so.0.2.0
-r-xr-xr-x 1 root root 129K Jan  1  1970 libusb-1.0.so.0.2.0
```
However
```python
>>> import os
>>> os.environ['LD_LIBRARY_PATH']
'/nix/store/z1j25bh18iaan59r4s8dn49lak06a582-libusb-1.0.23/lib/'
>>> import pyftdi.usbtools
>>> pyftdi.usbtools.UsbTools.find_all([(0x0403, 0x6014)])
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/nix/store/0g3g4a055bwpirvw70sz9z4095cj2zmb-python3-3.8.5-env/lib/python3.8/site-packages/pyftdi/usbtools.py", line 115, in find_all
    devs.update(UsbTools._find_devices(vid, pid, nocache))
  File "/nix/store/0g3g4a055bwpirvw70sz9z4095cj2zmb-python3-3.8.5-env/lib/python3.8/site-packages/pyftdi/usbtools.py", line 616, in _find_devices
    backend = cls._load_backend()
  File "/nix/store/0g3g4a055bwpirvw70sz9z4095cj2zmb-python3-3.8.5-env/lib/python3.8/site-packages/pyftdi/usbtools.py", line 683, in _load_backend
    raise ValueError('No backend available')
ValueError: No backend available
```
Running the same snippet on my laptop (an x86 system) returns the expected device descriptor.

---

It seems that `LD_LIBRARY_PATH` is simply ignored on the RPi; but I cannot find any usage of it in the code base to inspect.

eblot, 2020-11-10:
PyFtdi does not load the native libusb library on its own, this is handled by PyUSB.

PyUSB uses the `ctypes` module to load the native library, through its `find_library` method:

> ctypes.util.find_library(name)
> ...
> The exact functionality is system dependent.
> On Linux, find_library() tries to run external programs (/sbin/ldconfig, gcc, objdump and ld) to find the library file. It returns the filename of the library file.
> Changed in version 3.6: On Linux, the value of the environment variable LD_LIBRARY_PATH is used when searching for libraries, if a library cannot be found by any other means.

It should work with Python 3.8.5 AFAICT...

Native library loading is quite circumvolved. I do not have time for now to setup a matching test environment, could you have a look at pyusb's `liblibrary` module to see if anything unexpected is invoked?

tmplt, 2020-11-11:
Note to self: pyftdi and deps where here packaged by pypi2nix. pyftdi is otherwise packaged (likely more proper) in nixpkgs. Try to repro.

mcuee, 2021-08-22:
@tmplt Try the tips here.
https://github.com/pyusb/pyusb/blob/master/docs/tutorial.rst#specifying-libraries-by-hand