eblot/pyftdi #230: No support for I2C with FT2232D

dnltz, 2021-01-07:
Hey,

we have an old hardware and reworking our software for it. One part of that process is to switch to Python and use this package. The hardware has a FT2232D and I2C-GPIO-Expanders connected to it. Unfortunately, I2cController does not support this FTDI part because it's not from the H series and does not support 3 phase clocking: https://github.com/eblot/pyftdi/blob/650b6cb5f89d90c2321ee4d5ff0cfb350e0418ed/pyftdi/ftdi.py#L1559

Well, if we remove https://github.com/eblot/pyftdi/blob/650b6cb5f89d90c2321ee4d5ff0cfb350e0418ed/pyftdi/i2c.py#L501 manually, I2cController works and we can communicate with the expanders.

Can we enable this feature only on the H series?

eblot, 2021-01-07:
Did you validate the bus signal w/ an oscilloscope/logic analyzer? I have not been working with the FT2232D for many years, but I'm not sure the signals are valid w/o this feature.

dnltz, 2021-01-07:
The document [1] shows the differences on page 18. Looks to me like without 3 phase clocking, it's not implemented after i2c spec. However, we use a PCA9505 (NXP) which needs 0 ns between SDA changes and the falling edge.

So, maybe you have a good idea how we could disable the 3 phase clock without changing the code for all FTDIs.

1: https://www.ftdichip.com/Support/Documents/AppNotes/AN_355_FT232H%20MPSSE%20Example%20-%20I2C%20Master%20Interface%20with%20Visual%20Basic.pdf

eblot, 2021-01-07:
I guess we could add an *explicit* option (API) to force I2C mode with unsupported device.
This way, if someone wants to use this feature despite it generates weird signal, it is possible, but no one loses its time trying to figure out why it does not work w/o some kind of warning.

dnltz, 2021-01-07:
Sounds good. A colleague of mine will create a PR soon. Thanks for your fast reply!

eblot, 2021-01-08:
It needs to be something much more explicit and foolproof.

ie. something like
```py
def force_clock(self, enable: bool): -> None:
    """Force unsupported I2C clock signalling on devices that have no I2C capabilities (i.e. FT2232D).
    I2cController cowardly refuses to use unsupported devices.
    When this mode is enabled, I2cController can drive such devices, but I2C signalling is not compliant with I2C
    specifications and may not work with most I2C slaves.
    This is a fully unsupported feature (bug reports will be ignored)
 
    :param enable: whether to drive non-I2C capable devices.
    """
```

meuren, 2021-01-08:
Alternatively, a solution similar to this approach is possible:

```python
if self._force_clock_mode and not threephaseclock:
    pass
else:
    self._ftdi.enable_3phase_clock(True)`
```
which requires the following procedure
```python
>>> i2c = I2cController()
>>> i2c.force_clock_mode(True)
>>> i2c.configure(<url>, threephaseclock=False)
```

dnltz, 2021-01-11:
@meuren  - I think the threephaseclock parameter and the force_clock_mode is too much. Calling the force_clock_mode method should be enough to make clear this is a workaround.

eblot, 2021-01-11:
Please cross check v0.52.9 and let me know if it works with your project. Thanks.

meuren, 2021-01-11:
I tested our hardware with the version v0.52.9 and it worked properly. I my opinion this Issue can be closed, what do you think? @dnltz

dnltz, 2021-01-13:
@eblot - Thanks for your super fast support on this issue and the release!

closed 2021-01-13 by dnltz