eblot/pyftdi #236: F232H - Lock on State High for CBUS pins

kelyrr, 2021-03-04:
Hello,

Hello

With FTPROG i have switched the cbus_0 on "DRIVE1" and the LED turned on.
Then i switched to "DRIVE0" but the LED stays into HIGH state.

Please could you explain why this is happening ? What should i do ?

Also, i have difficult to understand what i should put into theses functions to control the GPIOs:
[image]

Thanks a lot for your help!! :)
Clément

eblot, 2021-05-06:
Whenever you update the EEPROM, you need to power cycle the device so that the EEPROM content is read and used.

Usage example on an UM232H board (FT232H module):

1. Load the current EEPROM content into a file
    `pyftdi/bin/ftconf.py -P 15ba:002a -o ft232h.ini ftdi://:232h/1`

2. Create a backup file if something goes wrong
    `cp ft232h.ini ft232h_backup.ini`

3. Edit the INI file
    ```ini
    [values]
    ; ...
    cbus_func_8 = DRIVE0
    cbus_func_9 = DRIVE1
    ```

4. Update the EEPROM
    `pyftdi/bin/ftconf.py  -i ft232h.ini -l values -u ftdi://:232h/1`

5. Unplug/plug back the UM232H module
    * yellow led lights up, green led is off

6. Edit the INI file
    ```ini
    [values]
    ; ...
    cbus_func_8 = DRIVE1
    cbus_func_9 = DRIVE0
    ```

7. Update the EEPROM
    `pyftdi/bin/ftconf.py  -i ft232h.ini -l values -u ftdi://:232h/1`

8. Unplug/plug back the UM232H module
    * green led lights up, yellow led is off

9. Edit the INI file
    ```ini
    [values]
    ; ...
    cbus_func_8 = GPIO
    cbus_func_9 = GPIO
    ```

10. Unplug/plug back the UM232H module
    * Now the module is ready to accept CBUS commands, so that AC8 and AC9 output can be changed dynamically
 
11. Run a small Python test
     On FT232H, CBUS drivable pins are ACBUS5, ACBUS6,ACBUS8, ACBUS9, from FT232H datasheet, therefore:
     * AC5 is b0, AC6 is b1, AC8 is b3, AC9 is b4.
     * to use AC8 and AC9, the mask is therefore 0b1100 or 0x0c; to use GPIO as output, the direction is 0xc as well 
    ```py
    from time import sleep
    from pyftdi.ftdi import Ftdi, FtdiError
    from pyftdi.eeprom import FtdiEeprom
   
    ftdi = Ftdi()
    ftdi.open_from_url(self.url)
    # sanity check: device should support CBUS feature
    self.assertEqual(ftdi.has_cbus, True)
    eeprom = FtdiEeprom()
    eeprom.connect(ftdi)
    print(f'{eeprom.cbus_mask:01x}')
    # sanity check: device should have been configured for CBUS GPIOs
    self.assertEqual(eeprom.cbus_mask & 0xc, 0xc)
    # configure CBUS8 and CBUS9 as output
    ftdi.set_cbus_direction(0xc, 0xc)
    loop = 10
    # alternate led blinking
    while loop:
        loop =- 1
        ftdi.set_cbus_gpio(0x8)
        sleep(0.2)
        ftdi.set_cbus_gpio(0x4)
        sleep(0.2)
    ftdi.close()
    ```

jbkroner, 2021-07-26:
I don't have any questions.  I just want to thank you for the great example you've left in this thread @eblot! 👏

Zibri, 2022-03-14:
> ```python
> ftdi.set_cbus_gpio(0x8)
> ```

My FT232R is configured in this way:
[image]

But,
>>> eeprom.cbus_mask
0

What should I do to access the 2 LEDS from python?