eblot/pyftdi #238: UART line setting constants differ from PySerial?

thestumbler, 2021-03-23:
I noticed that the line settings for parity, stop bits, and data size are different between pyserial and the pyftdi extension. Let's just use parity as an example...

```
import serial
port = serial.Serial('/dev/cu.usbserial-14720', 115200)
```
In pyserial, there are two flavors of parity constants. Flavor 1, which are lists of allowed parameters:
```
>>> dir(port)
['BAUDRATES', 'BAUDRATE_CONSTANTS', 'BYTESIZES', 'PARITIES', 'STOPBITS', ... etc ...
>>> port.PARITIES
('N', 'E', 'O', 'M', 'S')
```
Flavor 2, which are the individual values themselves:
```
>>> dir(serial)
[ ...snip... 'PARITY_EVEN', 'PARITY_MARK', 'PARITY_NAMES', 'PARITY_NONE', 'PARITY_ODD', 'PARITY_SPACE', ...etc...
```
But the same port via pyftdi extension, I can no longer get to the Flavor 2 constants:

```
import pyftdi.serialext
port = pyftdi.serialext.serial_for_url('ftdi://ftdi:232h:14:1c/1', 115200)

>>> dir(port) # this looks similar to pyserial, PARITIES is available
['BACKEND', 'BAUDRATES', 'BYTESIZES', 'PARITIES', 'PYSERIAL_VERSION', 'STOPBITS', ...etc...

>>> type(ser)
<class 'pyftdi.serialext.protocol_ftdi.Serial'>

>>> dir(pyftdi.serialext.protocol_ftdi.Serial)
['BACKEND', 'BAUDRATES', 'BYTESIZES', 'PARITIES', 'PYSERIAL_VERSION', ...etc...
# hmmm, PARITY_EVEN is missing...
>>> dir(pyftdi)
['FtdiLogger', 'NullHandler', 'WARNING', '__author__', ...etc...
>>> dir(pyftdi.serialext)
['SerialException', 'VERSION', '__builtins__', '__cached__',
>>> dir(pyftdi.serialext.protocol_ftdi)
['Ftdi', 'FtdiSerial', 'RawIOBase', 'Serial', 'SerialBase', 'SerialException', ...etc...
```
Question: Are these settings, like PARITY_EVEN, really there and I can't find them? Or has their use been deprecated? If I have a program and swap back and forth from pyserial to pyftdi, I have to code the line settings differently. Is this the intended behavior?

eblot, 2021-03-26:
`PARITY_*` are defined in PySerial module, that is in `serial.serial_util.py` which is imported when the serial module is imported.

If you want to list/use them, you need to import them, for example as `from serial.serialutil import *`. They are not automatically imported by PyFtdi because there are no dependency on these.

There are mere string definitions, which you can use as stated here: https://pythonhosted.org/pyserial/shortintro.html#configuring-ports-later

PyFtdi is compatible with these strings, e.g.  `pyftdi.serialext.serial_for_url('ftdi:///1', 115200, parity='E')`

If you do want to use the `PARITY_*` constants instead of their actual values, you need to import them.
```py
>>> from serial.serialutil import *
>>> from pyftdi import serialext
>>> port = serialext.serial_for_url('ftdi:///1', 115200, parity=PARITY_EVEN)
Serial<id=0x10ad07b80, open=True>(port='ftdi:///1', baudrate=115246, bytesize=8, parity='E', stopbits=1, timeout=None, xonxoff=False, rtscts=False, dsrdtr=False)
```

eblot, 2021-05-04:
No feedback.

Feel free to reopen if the above comment does not work as expected.

closed 2021-05-04 by eblot