eblot/pyftdi #239: FT2232H with 93C46 EEPROM: Invalid CRC or EEPROM content

eclispe, 2021-03-24:
Doesn't appear to take into account the chip field when calculating the EEPROM size.

If I edit eeprom.py to say that the 2232H has a size of 128bytes it parses correctly but obviously then fails to parse other devices with larger EEPROMs

Example dumped INI with size set to 128:
```
[values]
vendor_id = 0x0403
product_id = 0x6010
type = 0x0700
self_powered = True
remote_wakeup = False
power_max = 0
has_serial = True
suspend_pull_down = False
out_isochronous = False
in_isochronous = False
usb_version = 0x3333
manufacturer = FTDI
product = MiniTiouner-Express
serial = DA3OSHFP
channel_a_driver = D2XX
channel_b_driver = D2XX
group_0_drive = True
group_0_schmitt = False
group_0_slew = False
group_1_drive = True
group_1_schmitt = False
group_1_slew = False
group_2_drive = True
group_2_schmitt = False
group_2_slew = False
group_3_drive = True
group_3_schmitt = False
group_3_slew = False
chip = 0x46
channel_a_type = UART
channel_b_type = FIFO
suspend_dbus7 = Cfg1.0

[raw]
@00 = 0001030410600007c000080033339a0a
@10 = a428cc120000000046000a0346005400
@20 = 4400490028034d0069006e0069005400
@30 = 69006f0075006e00650072002d004500
@40 = 78007000720065007300730012034400
@50 = 410033004f0053004800460050000000
@60 = 00000000000000000000000000000000
@70 = 0000000000000000000000000000f56f
```

eblot, 2021-05-04:
Sorry,  I need more details here:

* what did you run?
* what do you expect
* what is the actual size of the EEPROM you use (or its type)
* what is the `chip field` you are referring to in this case?

eblot, 2021-05-04:
On my FT2232H mini module, the EEPROM is 256 byte long and parsed properly.

I understand you are using a FT2232H device with a 128 byte EEPROM?

PyFtdi does not implement a heuristic to detect the size of the EEPROM, maybe this is what you meant?

eblot, 2021-05-04:
Ok, I got what you meant, PyFtdi does not use the `chip` information to adjust the actual EEPROM size to the EEPROM model.
I'll fix that.

eclispe, 2021-05-04:
Yes, that is what I meant, I should have been more clear in the report.
I have boards with both sizes of EEPROM here if you need testing.

eblot, 2021-05-05:
It will be quite difficult to test with a real HW... Is it a board available for purchase somewhere or a custom design?

eclispe, 2021-05-05:
The problem was discovered on a MiniTiouner-Express and I have one on loan, they are available from https://www.datv-express.com/ for USD $75 plus shipping.
Would flashing a standard FT2232H module with the incorrect chip ID work for testing?

eblot, 2021-05-05:
It would help but I'm pretty sure there would be several corner cases that would fail or even worse I would not add regressions...

I've found a batch of 93C46/56/66/... EEPROMs for a couple of € and ordered them. They will not be delivered till end of may though. I will try to create a couple of HW setup to test various FTDI/EEPROM pairs and try to improve EEPROM handling with these various combinations.

Meanwhile, I've added an additional check based on EEPROM content: 31da7fc9f. I'm not sure if it can help in your case. Leet me know.

eclispe, 2021-05-05:
That is certainly an improvement, it passes the CRC now and dumps most of the data but isn't doing the strings.
```
[values]
vendor_id = 0x0403
product_id = 0x6010
type = 0x0700
self_powered = True
remote_wakeup = False
power_max = 0
has_serial = True
suspend_pull_down = False
out_isochronous = False
in_isochronous = False
usb_version = 0x3333
manufacturer =
product =
serial =
channel_a_driver = D2XX
channel_b_driver = D2XX
group_0_drive = True
group_0_schmitt = False
group_0_slew = False
group_1_drive = True
group_1_schmitt = False
group_1_slew = False
group_2_drive = True
group_2_schmitt = False
group_2_slew = False
group_3_drive = True
group_3_schmitt = False
group_3_slew = False
chip = 0x46
channel_a_type = UART
channel_b_type = FIFO
suspend_dbus7 = Cfg1.0

[raw]
@00 = 0001030410600007c000080033339a0a
@10 = a428cc120000000046000a0346005400
@20 = 4400490028034d0069006e0069005400
@30 = 69006f0075006e00650072002d004500
@40 = 78007000720065007300730012034400
@50 = 410033004f0053004800460050000000
@60 = 00000000000000000000000000000000
@70 = 0000000000000000000000000000f56f
```

eblot, 2021-05-05:
Ok, I'll have a look at it. Thanks for the feedback.

eblot, 2021-05-06:
9e4f7b82c should address this issue.

Note that I've not run many regression tests on this, so use with care if you plan to update the EEPROM content, as usual.

eclispe, 2021-05-06:
Excellent, thanks for sorting it.
That appears to work with ftconf, I'll do some further testing with the rest of my code to check for regressions.
```
[values]
chip = 0x46
vendor_id = 0x0403
product_id = 0x6010
type = 0x0700
self_powered = True
remote_wakeup = False
power_max = 0
has_serial = True
suspend_pull_down = False
out_isochronous = False
in_isochronous = False
usb_version = 0x3333
manufacturer = FTDI
product = MiniTiouner-Express
serial = DA3OSHFP
channel_a_driver = D2XX
channel_b_driver = D2XX
group_0_drive = True
group_0_schmitt = False
group_0_slew = False
group_1_drive = True
group_1_schmitt = False
group_1_slew = False
group_2_drive = True
group_2_schmitt = False
group_2_slew = False
group_3_drive = True
group_3_schmitt = False
group_3_slew = False
channel_a_type = UART
channel_b_type = FIFO
suspend_dbus7 = Cfg1.0

[raw]
@00 = 0001030410600007c000080033339a0a
@10 = a428cc120000000046000a0346005400
@20 = 4400490028034d0069006e0069005400
@30 = 69006f0075006e00650072002d004500
@40 = 78007000720065007300730012034400
@50 = 410033004f0053004800460050000000
@60 = 00000000000000000000000000000000
@70 = 0000000000000000000000000000f56f
```

eclispe, 2021-05-10:
Tested more thoroughly and it seems to work with my application

closed 2021-05-10 by eclispe

eblot, 2021-05-11:
Thanks for the feedback. Delivered as part of v0.53.1.