eblot/pyftdi #248: Error Message:pyftdi.ftdi.FtdiError: UsbError: [Errno 13] Access denied (insufficient permissions)

949699582, 2021-04-24:
Hello eblot:
I'm using this pyftdi library for a professionnal project.
It is very great. Thanks you for that, it's a big help !

But I meet an error when I use it.The error info is "pyftdi.ftdi.FtdiError: UsbError: [Errno 13] Access denied (insufficient permissions)".
Operating environment:MacOS Mojave 
version:10.14.6

It worked fine when I ran the code on the Mac of the newly installed system!
But when I run the code on an old computer, it gets an error!
I looked at the official documentation describing this.But I still don't know how to do it. I apply for an official answer here!

Best Regards!

Here is the error detail:

    self.i2cSlave.configure(portname,frequency=400000)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/pyftdi/i2c.py", line 498, in configure
    frequency = self._ftdi.open_mpsse_from_url(url, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/pyftdi/ftdi.py", line 632, in open_mpsse_from_url
    debug=debug)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/pyftdi/ftdi.py", line 750, in open_mpsse_from_device
    frequency = self._set_frequency(frequency)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/pyftdi/ftdi.py", line 2288, in _set_frequency
    self.write_data(cmd)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/pyftdi/ftdi.py", line 1757, in write_data
    length = self._write(data[offset:offset+write_size])
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/pyftdi/ftdi.py", line 2068, in _write
    raise FtdiError('UsbError: %s' % str(ex)) from None
pyftdi.ftdi.FtdiError: UsbError: [Errno 13] Access denied (insufficient permissions)

eblot, 2021-04-24:
Have you tried https://eblot.github.io/pyftdi/troubleshooting.html#error-access-denied-insufficient-permissions

eblot, 2021-05-04:
No feedback.

Feel free to re-open if the above suggestion does not work for you.

closed 2021-05-04 by eblot