eblot/pyftdi #256: ftdi_urls.py crashes with pyusb v1.2.0

obpt, 2021-07-05:
Good evening,

With pyusb v1.2.0, the script `ftdi_urls.py` crashes with the following error:

```
Traceback (most recent call last):
  File "[...]/pyftdi/pyftdi/bin/ftdi_urls.py", line 74, in <module>
    main()
  File "[...]/pyftdi/pyftdi/bin/ftdi_urls.py", line 62, in main
    Ftdi.show_devices()
  File "[...]/.venv/lib/python3.9/site-packages/pyftdi/ftdi.py", line 387, in show_devices
    devdescs = UsbTools.list_devices(url or 'ftdi:///?',
  File "[...]/.venv/lib/python3.9/site-packages/pyftdi/usbtools.py", line 274, in list_devices
    candidates, _ = cls.enumerate_candidates(urlparts, vdict, pdict,
  File "[...]/.venv/lib/python3.9/site-packages/pyftdi/usbtools.py", line 417, in enumerate_candidates
    devices = cls.find_all(vps)
  File "[...]/.venv/lib/python3.9/site-packages/pyftdi/usbtools.py", line 95, in find_all
    devs.update(UsbTools._find_devices(vid, pid, nocache))
  File "[...]/.venv/lib/python3.9/site-packages/pyftdi/usbtools.py", line 615, in _find_devices
    devs.add(device)
TypeError: unhashable type: 'Device'
```
With pyusb v1.1.1, no crash occurs.

Regards,

obpt

eblot, 2021-07-08:
Thanks for catching and reporting this issue.

UsbTools is a very old piece of code that needs a full rewrite, although PyUSB should maintain backward compatibility. Meanwhile, I've disabled support for PyUSB 1.2.0 in PyFtdi v0.53.2, till PyFtdi gets some major overhaul with UsbTools.

mcuee, 2021-07-09:
FYI, pyusb-1.2.1 has been released.

mcuee, 2021-08-22:
It will be good to unblock pyusb-1.2.1 if possible.

eblot, 2021-08-22:
> It will be good to unblock pyusb-1.2.1 if possible.

Could you give a try to 6a3e68dc6. If it is ok, I will tag it. Thanks

mcuee, 2021-08-22:
Looks good to me.

```

pyftdi on  master via 🐍 v2.7.16 ❯ pip3 install .         
DEPRECATION: Configuring installation scheme with distutils config files is deprecated and will no longer work in the near future. If you are using a Homebrew or Linuxbrew Python, please see discussion at https://github.com/Homebrew/homebrew-core/issues/76621
Processing /Users/mcuee/build/libftdi/pyftdi
  DEPRECATION: A future pip version will change local packages to be built in-place without first copying to a temporary directory. We recommend you use --use-feature=in-tree-build to test your packages with this new behavior before it becomes the default.
   pip 21.3 will remove support for this functionality. You can find discussion regarding this at https://github.com/pypa/pip/issues/7555.
Collecting pyusb!=1.2.0,>=1.0.0
  Using cached pyusb-1.2.1-py3-none-any.whl (58 kB)
Collecting pyserial>=3.0
  Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)
Building wheels for collected packages: pyftdi
  Building wheel for pyftdi (setup.py) ... done
  Created wheel for pyftdi: filename=pyftdi-0.53.2-py3-none-any.whl size=141544 sha256=4304e1fd5d143c831df1b62c4a7d64193f1291f363c3b600356e89f7e7dd67e3
  Stored in directory: /private/var/folders/zf/457r6vn50knfpnjgy1156llr0000gn/T/pip-ephem-wheel-cache-rcbhpngr/wheels/10/d9/3d/29da3e0fbd69ee5bc2080220f8e57fd9e7bf5ad3ee8b2daf4d
Successfully built pyftdi
Installing collected packages: pyusb, pyserial, pyftdi
  DEPRECATION: Configuring installation scheme with distutils config files is deprecated and will no longer work in the near future. If you are using a Homebrew or Linuxbrew Python, please see discussion at https://github.com/Homebrew/homebrew-core/issues/76621
  DEPRECATION: Configuring installation scheme with distutils config files is deprecated and will no longer work in the near future. If you are using a Homebrew or Linuxbrew Python, please see discussion at https://github.com/Homebrew/homebrew-core/issues/76621
  DEPRECATION: Configuring installation scheme with distutils config files is deprecated and will no longer work in the near future. If you are using a Homebrew or Linuxbrew Python, please see discussion at https://github.com/Homebrew/homebrew-core/issues/76621
DEPRECATION: Configuring installation scheme with distutils config files is deprecated and will no longer work in the near future. If you are using a Homebrew or Linuxbrew Python, please see discussion at https://github.com/Homebrew/homebrew-core/issues/76621
Successfully installed pyftdi-0.53.2 pyserial-3.5 pyusb-1.2.1
```

mcuee, 2021-08-23:
Last test was done on my Mac Mini M1 (macOS Big Sur ARM64)
Also good under Windows 10 x64.
```
(py39venv) C:\work\libusb\pyftdi [master ≡]> pip install .
Processing c:\work\libusb\pyftdi
  DEPRECATION: A future pip version will change local packages to be built in-place without first copying to a temporary directory. We recommend you use --use-feature=in-tree-build to test your packages with this new behavior before it becomes the default.
   pip 21.3 will remove support for this functionality. You can find discussion regarding this at https://github.com/pypa/pip/issues/7555.
Collecting pyusb!=1.2.0,>=1.0.0
  Using cached pyusb-1.2.1-py3-none-any.whl (58 kB)
Collecting pyserial>=3.0
  Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)
Building wheels for collected packages: pyftdi
  Building wheel for pyftdi (setup.py) ... done
  Created wheel for pyftdi: filename=pyftdi-0.53.2-py3-none-any.whl size=141529 sha256=2195d6f4cc206e085e26ee497931bae3f7174f8aff513a77044f4afed993c459
  Stored in directory: C:\Users\xfchen\AppData\Local\Temp\pip-ephem-wheel-cache-0559ahm0\wheels\12\24\bd\af8146c8e6a65dabbea6724dcc35b3ba700fc77338c6856517
Successfully built pyftdi
Installing collected packages: pyusb, pyserial, pyftdi
Successfully installed pyftdi-0.53.2 pyserial-3.5 pyusb-1.2.1
```

eblot, 2021-08-23:
Thanks a lot for the feedback. v0.53.3 is tagged and available on PyPI.

closed 2021-08-23 by eblot