eblot/pyftdi #257: unhashable type: 'Device'

forestomir, 2021-07-07:
while executing "ftdi_urls.py" i get following error:

`
Traceback (most recent call last):
  File "/home/bob/.local/bin/ftdi_urls.py", line 74, in <module>
    main()
  File "/home/bob/.local/bin/ftdi_urls.py", line 62, in main
    Ftdi.show_devices()
  File "/home/bob/.local/lib/python3.8/site-packages/pyftdi/ftdi.py", line 387, in show_devices
    devdescs = UsbTools.list_devices(url or 'ftdi:///?',
  File "/home/bob/.local/lib/python3.8/site-packages/pyftdi/usbtools.py", line 274, in list_devices
    candidates, _ = cls.enumerate_candidates(urlparts, vdict, pdict,
  File "/home/bob/.local/lib/python3.8/site-packages/pyftdi/usbtools.py", line 417, in enumerate_candidates
    devices = cls.find_all(vps)
  File "/home/bob/.local/lib/python3.8/site-packages/pyftdi/usbtools.py", line 95, in find_all
    devs.update(UsbTools._find_devices(vid, pid, nocache))
  File "/home/bob/.local/lib/python3.8/site-packages/pyftdi/usbtools.py", line 620, in _find_devices
    devs.add(device)
TypeError: unhashable type: 'Device'
`

my setup is:
ubuntu 20.04
Python 3.8.10 (default on ubuntu)
pyftdi 0.53.1 (installed by pip)
pyusb 1.2.0 (installed by pip to fulfill dependencies)

forestomir, 2021-07-07:
I have just noticed that yesterday same issue was reported under
"ftdi_urls.py crashes with pyusb v1.2.0 #256"
so this one can be marked as duplicated

eblot, 2021-07-13:
Dup of #256

closed 2021-07-13 by eblot