eblot/pyftdi #259: "The device has no langid" on Win10

forestomir, 2021-07-13:
i run REPL as administrator and get following

`
Python 3.9.6 (tags/v3.9.6:db3ff76, Jun 28 2021, 15:26:21) [MSC v.1929 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> from pyftdi.i2c import I2cController
>>> i2c=I2cController()
>>> i2c.ftdi.list_devices()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\bob\AppData\Local\Programs\Python\Python39\lib\site-packages\pyftdi\ftdi.py", line 375, in list_devices
    return UsbTools.list_devices(url or 'ftdi:///?',
  File "C:\Users\bob\AppData\Local\Programs\Python\Python39\lib\site-packages\pyftdi\usbtools.py", line 274, in list_devices
    candidates, _ = cls.enumerate_candidates(urlparts, vdict, pdict,
  File "C:\Users\bob\AppData\Local\Programs\Python\Python39\lib\site-packages\pyftdi\usbtools.py", line 417, in enumerate_candidates
    devices = cls.find_all(vps)
  File "C:\Users\bob\AppData\Local\Programs\Python\Python39\lib\site-packages\pyftdi\usbtools.py", line 101, in find_all
    description = UsbTools.get_string(dev, dev.iProduct)
  File "C:\Users\bob\AppData\Local\Programs\Python\Python39\lib\site-packages\pyftdi\usbtools.py", line 556, in get_string
    return usb_get_string(device, stridx)
  File "C:\Users\bob\AppData\Local\Programs\Python\Python39\lib\site-packages\usb\util.py", line 313, in get_string
    raise ValueError("The device has no langid"
ValueError: The device has no langid (permission issue, no string descriptors supported or device error)
`

I had similar problem on ubuntu but simple udev rule, expanding access permissions, did the job
`$ cat /etc/udev/rules.d/FTDI.rules 
SUBSYSTEMS=="usb", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6011", MODE="0666"`

On Windows I can not do that, hoped that running REPL as Administrator would help but it did not. Please advise

my setup is:
`
C:\>pip.exe show pyftdi
Name: pyftdi
Version: 0.53.2
Summary: FTDI device driver (pure Python)
Home-page: http://github.com/eblot/pyftdi
Author: Emmanuel Blot
Author-email: emmanuel.blot@free.fr
License: Modified BSD
Location: c:\users\bob\appdata\local\programs\python\python39\lib\site-packages
Requires: pyusb, pyserial
Required-by:

C:\>pip.exe show pyusb
Name: pyusb
Version: 1.1.0
Summary: Python USB access module
Home-page: https://pyusb.github.io/pyusb
Author: Robert Wlodarczyk
Author-email: robert@simplicityguy.com
License: UNKNOWN
Location: c:\users\bob\appdata\local\programs\python\python39\lib\site-packages
Requires:
Required-by: pyftdi
`

eblot, 2021-07-13:
> I had similar problem on ubuntu but simple udev rule, expanding access permissions, did the job
> `$ cat /etc/udev/rules.d/FTDI.rules SUBSYSTEMS=="usb", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6011", MODE="0666"`

Yes, this is what is described in the [installation section](https://eblot.github.io/pyftdi/installation.html#debian-ubuntu-linux) of the documentation.

> On Windows I can not do that, hoped that running REPL as Administrator would help but it did not. Please advise

No [idea](https://eblot.github.io/pyftdi/index.html#supported-host-oses). Windows is a plague, and a bag of sh*t when it comes to USB. Did you follow the [Windows installation instructions](https://eblot.github.io/pyftdi/installation.html#windows)?.

forestomir, 2021-07-13:
thanks @eblot for tip with Zadig, Unfortunatelly It didnt work for me, and for now my solution is to use ftd2xx instead of pyftdi

closed 2021-07-13 by forestomir