eblot/pyftdi #269: How to write VID:PID?

dattasaurabh82, 2021-08-31:
**OS:** macOS Mojave 10.14.6
**Testing Target chip:** FT232R
**Future Target chip:** FT232RQ

I managed to write: to EEPROM
1. `manufacturer`
2. `product_name`
3. `serial_number` 
with:
```
ftconf.py ftdi://ftdi:232:12345678/1 -m <custom_name> -p <custom_name>  -s <custom_ser_no.> -u -v
```

Now I want to write **custom VID and PID**. 

_Note: I'm aware that if that is successful, I'm on my own to make the driver load the new set._ 

Say I have:
1. VID: [Decimal] 11219 [Hex] 0x2BD3
2. PID: [Decimal] 22979 [Hex] 0x59C3

After reading the docu, I assume the command would be:  

```
ftconf.py ftdi://ftdi:232:12345678/1 --vidpid 0x2BD3:0x59C3 -u -v
```
But this doesn't write those VIDs and PIDs. (As the device is still available to be read from with FTDI's default VID:PID)
Am I missing somethings ? 

Can we do this at-all from mac or Linux or need to use FTProg or D2XX from Windows?

eblot, 2021-09-03:
There is a misunderstanding about the role of `--vidpid`. It does not change anything in the EEPROM. It only tells PyFdti what non-standard VID/PID pairs should be considered as FTDI devices. In other words, this enables you to access the PyFdti device **once** its EEPROM has been customized with non-standard VID/PID - otherwise the device would not be recognized at all.

What you want to do is:
1. Save the current EEPROM content into an editable file:
    ```sh
    pyftdi/bin/ftconf.py ftdi:///1 -o mydevice.ini
    ```
2. Edit the configuration file to assign the new VID/PID
    ```diff
    --- a/mydevice.ini
    +++ b/mydevice.ini
    @@ -1,6 +1,6 @@
     [values]
    -vendor_id = 0x0403
    -product_id = 0x6001
    +vendor_id = 0x2BD3
    +product_id = 0x59C3
     type = 0x0600
     self_powered = false
     remote_wakeup = true
     ```
 3. Update the EEPROM with the new configuration
     ```sh
    pyftdi/bin/ftconf.py ftdi:///1 -i mydevice.ini -u
    ```
 4. Power-cycle the device so that it reads its config
    ```sh
    lsusb
    ```
    ```
    Bus 002 Device 004: ID 2bd3:59c3 2bd3 UM232R USB <-> Serial  Serial: FTGXSYWJ
    ```
5. Access the updated device with the new VID/PID
    ```sh
    pyftdi/bin/ftconf.py ftdi:///1 --vidpid 0x2BD3:0x59C3 -v
    ```
    ```
    vendor_id: 0x2bd3
    product_id: 0x59c3
    type: 0x0600
    ...
    ```
6. (Opt.) Cross-check that the device is no longer listed as a regular FTDI device
    ```sh
    pyftdi/bin/ftconf.py ftdi:///1 -v
    ```
    ```
    pyftdi.usbtools.UsbToolsError: No USB device matches URL ftdi:///1
    ```

Hope it clarifies the whole procedure.
I ran it on macOS, but it should be the same on any Linux host.
On Windows, I have no idea (and to be honest ... I do not care about this crappy OS), I guess that you need to update the Zadig configuration at some stage.

PyFdti has no relationship with FTProg or any FTDI DLL, so they are never used nor even checked.

dattasaurabh82, 2021-09-06:
One Question, for updating the new config file (.ini), does the `[raw]` have any affect? 

I'm asking because, if I change the `key:value` pairs under `[values]`, do I also have to modify the hex section as I imagine (might be wrong here), the `[raw]` section help set up the editable/human readable `[values]` section (but they are basically the same)?

eblot, 2021-09-06:
As stated in https://eblot.github.io/pyftdi/eeprom.html#option-switches `[values]` section takes precedence over `[raw]` :-)

eblot, 2021-09-06:
The `[raw]` section may be used to customize parts of the EEPROM that are not supported by standard options - or user-section data. It may also be used for copying data from one device to another, etc.

There is an option switch to tell ftconf which section to use, in case you want to ignore one section w/o removing it from the file.

By default, raw section is read then it is overridden w/ data from the values section. This ensures that unsupported parts of EEPROM are not replaced with a default, meaningless bit sequence.

dattasaurabh82, 2021-09-07:
Thanks for the great explanation

closed 2021-09-07 by dattasaurabh82