eblot/pyftdi #278: group_(0|1)_(drive|schmitt|slow_slew) option causes error

aeular, 2021-12-14:
When trying to check or set group_(0|1)_(drive|schmitt|slow_slew) on an FT232H throws the error:

```
$ python3 ftconf.py ftdi:///1 -c group_0_drive=?
Traceback (most recent call last):
  File "/usr/local/bin/ftconf.py", line 158, in <module>
    main()
  File "/usr/local/bin/ftconf.py", line 124, in main
    eeprom.set_property(name, value, helpio)
  File "/usr/local/lib/python3.9/dist-packages/pyftdi/eeprom.py", line 397, in set_property
    self._set_group(int(mobj.group(1)), mobj.group(2), value, out)
TypeError: _set_group() takes 4 positional arguments but 5 were given
```

```
$ python3 ftconf.py ftdi:///1 -c group_0_drive=8
Traceback (most recent call last):
  File "/usr/local/bin/ftconf.py", line 158, in <module>
    main()
  File "/usr/local/bin/ftconf.py", line 124, in main
    eeprom.set_property(name, value, helpio)
  File "/usr/local/lib/python3.9/dist-packages/pyftdi/eeprom.py", line 397, in set_property
    self._set_group(int(mobj.group(1)), mobj.group(2), value, out)
TypeError: _set_group() takes 4 positional arguments but 5 were given
```

Possible options:
  ```
$ python3 ftconf.py ftdi:///1 -c \?
cbus_func_0, cbus_func_1, cbus_func_2, cbus_func_3, cbus_func_4,
  cbus_func_5, cbus_func_6, cbus_func_7, cbus_func_8, cbus_func_9,
  channel_a_driver, channel_a_type, chip, clock_polarity,
  flow_control, group_0_drive, group_0_schmitt, group_0_slow_slew,
  group_1_drive, group_1_schmitt, group_1_slow_slew, has_serial,
  in_isochronous, lsb_data, out_isochronous, power_max, powersave,
  product_id, remote_wakeup, self_powered, suspend_pull_down, type,
  vendor_id
```

Possibly related, but changing the values in the ini file and updating doesn't appear to change the channel_(0|1)_drive either (trying to go from 4 mA to 8 mA, when it's allowed 4/8/12/16)

AndreRenaud, 2022-03-07:
I had this same issue. The fix for me appears to be to add the extra `control` argument to `_set_group`, and then pass it through to `_set_group_x232h` (in `eeprom.py`)
```python
    def _set_group(self, group: int, control: str, value: Union[str, int, bool],
                   out: Optional[TextIO]) -> None:
        if self.device_version in (0x0700, 0x0800, 0x0900):
            self._set_group_x232h(group, control, value, out)
            return
        raise ValueError('Group not implemented for this device')
```

aeular, 2022-03-07:
Thanks for that, that for sure let me increase the drive current from 4 mA to 8 mA on linux, hopefully this gets fixed in the repo

eblot, 2022-03-09:
Yes, very sorry about the delay, I'll try to update the repo soon.

commit 3b76d8e7 (2022-03-13): Fix #278: missing `control` value in `_set_group`
closed 2022-03-13 by eblot