eblot/pyftdi #279: ftdi:ft-x - This interface does not support MPSSE

pacmac, 2021-12-20:
Hi, firstly thanks for developing this.

I'm obviously using this for the first time, and are attempting to test I2C communications.

It seems that the device is detected, but I am unable to configure i2c, with the error regarding MPSSE, is this something that can be configured or is it a compatibility issue ?

```
>>> Ftdi.show_devices()
Available interfaces:
  ftdi://ftdi:ft-x:D200B4HD/1   (FT200X USB I2C)

>>>
>>>
>>> i2c.configure('ftdi://ftdi:ft-x:D200B4HD/1')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.8/site-packages/pyftdi/i2c.py", line 498, in configure
    frequency = self._ftdi.open_mpsse_from_url(url, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/pyftdi/ftdi.py", line 635, in open_mpsse_from_url
    return self.open_mpsse_from_device(device, interface,
  File "/usr/local/lib/python3.8/site-packages/pyftdi/ftdi.py", line 737, in open_mpsse_from_device
    raise FtdiMpsseError('This interface does not support MPSSE')
pyftdi.ftdi.FtdiMpsseError: This interface does not support MPSSE
>>>
```

eblot, 2021-12-20:
Hi,

FT-200X is not an I2C master. It is a crap I2C slave, which acts like a sort of uber-limited bridge between an I2C bus and a USB bus: https://ftdichip.com/products/ft200xd/

It does not support the MPSEE protocol (used by FT-*232* devices) that PyFtdi support.
There is no plan to support FT-200X in PyFtdi, as it is not related to MPSEE, and its usage is more than limited, unfortunately.

Are you looking for an I2C master or an I2C slave device?

pacmac, 2021-12-20:
Thanks, that explains it, did not realise it was a slave only device. I was looking for a master to use to replace a solution that I designed some time ago on a micropython board so that I could run it from my mac instead.

I will order another device

Thanks again.

closed 2021-12-20 by pacmac

sifive-eblot, 2021-12-21:
The FT232H is the best if you need a single function at once, yes.

Beware that chinese vendors advertise for many devices at once, but you need to be very careful when you pick a "color", *i.e.* the actual model of a device.

A FT232H board is about €10 (+/- 1€ depending on the shipping fee).

FTDI devices are incredibly expensive... and not bug free :-)

pacmac, 2021-12-21:
Thanks, I already have ordered a few adafruit 2264 https://www.adafruit.com/product/2264 from digikey.

The listing I purchased from originally stating: "FT232H Multifunction High-Speed USB to JTAG UART/ FIFO SPI/ I2C CJMCU Module" - was a single listing, there was no option to select, price was about USD5.00, definitely a con.