eblot/pyftdi #285: FTDI - URL string is missing device port

RazKor, 2022-01-11:
API documentation » serialext - UART API

Using a device ft231x and trying to read data at baud rate of 1000000 , it does not work 

Using the 2 command lines 
import pyftdi.serialext
port = pyftdi.serialext.serial_for_url('ftdi://ftdi:ft231x', baudrate=1000000)

There is example of :
import pyftdi.serialext
port = pyftdi.serialext.serial_for_url('ftdi://ftdi:2232h/2', baudrate=3000000)

What Can be the problem ?

eblot, 2022-01-12:
Have you tried 

```py
import pyftdi.serialext
port = pyftdi.serialext.serial_for_url('ftdi:///?')
```

to get the actually supported URL strings?

eblot, 2022-03-13:
No feedback, closing. See Discussions to ask for help on how to use Pyftdi.

closed 2022-03-13 by eblot