eblot/pyftdi #286: Problem with an Nuvoton device part of an RND 320-KA3005P power supply

friedrichromstedt, 2022-01-14:
Today I struggled in communicating with an RND 320-KA3005P power supply.

I installed a proper udev rule, such that:
```
from pyftdi.ftdi import Ftdi

Ftdi.add_custom_vendor(0x0416, 'nuvoton')
Ftdi.add_custom_product(0x0416, 0x5011, 'usb-com')

Ftdi.show_devices()
```
replies by:
```
Available interfaces:
  ftdi://nuvoton:usb-com:002912C90241/1   (USB Virtual COM)
  ftdi://nuvoton:usb-com:002912C90241/2   (USB Virtual COM)
```
"Nuvoton" is the manufacturer's name, it shows up in the `journalctl` output when connecting the device via the USB port.

I am experiencing problems when attempting to open a communication connection.  I've tried the following commands:
```
ftdi = Ftdi()
ftid.open_from_url('ftdi://nuvoton:usb-com/1')
```
```
ftdi = Ftdi.create_from_url('ftdi://nuvoton:usb-com/1')
```
```
port = pyftdi.serialext.serial_for_url(
        'ftdi://nuvoton:usb-com/1',
        baudrate=9600, bytesize=8, parity='N', stopbits=1)
```
And the same with the other URL `(...)/2`.  The arguments to `pyftdi.serialext.serial_for_url` are the figures given in the manual.

The precise error messages are for the `(...)/1` URL, using `open_from_url`:
```
Traceback (most recent call last):
  File (xxx), line 7, in <module>
    ftdi.open_from_url('ftdi://nuvoton:usb-com/1')
  File "/usr/lib/python3.10/site-packages/pyftdi/ftdi.py", line 519, in open_from_url
    self.open_from_device(device, interface)
  File "/usr/lib/python3.10/site-packages/pyftdi/ftdi.py", line 572, in open_from_device
    self._set_interface(config, interface)
  File "/usr/lib/python3.10/site-packages/pyftdi/ftdi.py", line 2027, in _set_interface
    self._in_ep, self._out_ep = endpoints[:2]
ValueError: not enough values to unpack (expected 2, got 1)
```
and for the `(...)/2` URL, using `create_from_url`:
```
Traceback (most recent call last):
  File (xxx), line 6, in <module>
    ftdi1 = Ftdi.create_from_url('ftdi://nuvoton:usb-com/2')
  File "/usr/lib/python3.10/site-packages/pyftdi/ftdi.py", line 384, in create_from_url
    device.open_from_url(url)
  File "/usr/lib/python3.10/site-packages/pyftdi/ftdi.py", line 519, in open_from_url
    self.open_from_device(device, interface)
  File "/usr/lib/python3.10/site-packages/pyftdi/ftdi.py", line 578, in open_from_device
    self.purge_buffers()
  File "/usr/lib/python3.10/site-packages/pyftdi/ftdi.py", line 1170, in purge_buffers
    self.purge_rx_buffer()
  File "/usr/lib/python3.10/site-packages/pyftdi/ftdi.py", line 1154, in purge_rx_buffer
    if self._ctrl_transfer_out(Ftdi.SIO_REQ_RESET,
  File "/usr/lib/python3.10/site-packages/pyftdi/ftdi.py", line 2064, in _ctrl_transfer_out
    raise FtdiError('UsbError: %s' % str(ex)) from None
pyftdi.ftdi.FtdiError: UsbError: [Errno 110] Operation timed out
```
Using the more explicit statement involving `pyftdi.serialext` leads to the same error for the `(...)/1` URL:
```
Traceback (most recent call last):
  File (xxx), line 10, in <module>
    port = pyftdi.serialext.serial_for_url(
  File "/usr/lib/python3.10/site-packages/serial/__init__.py", line 90, in serial_for_url
    instance.open()
  File "/usr/lib/python3.10/site-packages/pyftdi/serialext/protocol_ftdi.py", line 50, in open
    device = Ftdi.create_from_url(self.port)
  File "/usr/lib/python3.10/site-packages/pyftdi/ftdi.py", line 384, in create_from_url
    device.open_from_url(url)
  File "/usr/lib/python3.10/site-packages/pyftdi/ftdi.py", line 519, in open_from_url
    self.open_from_device(device, interface)
  File "/usr/lib/python3.10/site-packages/pyftdi/ftdi.py", line 572, in open_from_device
    self._set_interface(config, interface)
  File "/usr/lib/python3.10/site-packages/pyftdi/ftdi.py", line 2027, in _set_interface
    self._in_ep, self._out_ep = endpoints[:2]
ValueError: not enough values to unpack (expected 2, got 1)
```
The `journalctl` output of connecting the device is:
```
Jan 14 12:37:31 T470p kernel: usb 1-2: new full-speed USB device number 7 using xhci_hcd
Jan 14 12:37:32 T470p kernel: usb 1-2: New USB device found, idVendor=0416, idProduct=5011, bcdDevice= 3.00
Jan 14 12:37:32 T470p kernel: usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jan 14 12:37:32 T470p kernel: usb 1-2: Product: USB Virtual COM
Jan 14 12:37:32 T470p kernel: usb 1-2: Manufacturer: Nuvoton
Jan 14 12:37:32 T470p kernel: usb 1-2: SerialNumber: 002912C90241
Jan 14 12:37:32 T470p kernel: cdc_acm 1-2:1.0: ttyACM0: USB ACM device
Jan 14 12:37:32 T470p kernel: usbcore: registered new interface driver cdc_acm
Jan 14 12:37:32 T470p kernel: cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
```
I searched the web, however it is difficult to find information about a tuple unpacking `ValueError`, even with some obvious keywords like `pyftdi` the search did not yield relevant results. I've tried the obvious steps of restarting and replugging the device.

This is with `pyftdi-0.52.0`.

eblot, 2022-01-14:
Try to use `ftdi://0x0416:0x5011:002912C90241/1`to check if that helps. There may be an issue with custom vendor string, I need to double check

friedrichromstedt, 2022-01-15:
Running:
```
from pyftdi.ftdi import Ftdi
import pyftdi.serialext

Ftdi.add_custom_vendor(0x0416)
Ftdi.add_custom_product(0x0416, 0x5011)

ftdi = Ftdi.create_from_url('ftdi://0x0416:0x5011:002912C90241/1')
```
results in the already kown Traceback:
```
Traceback (most recent call last):
  File (xxx), line 7, in <module>
    ftdi = Ftdi.create_from_url('ftdi://0x0416:0x5011:002912C90241/1')
  File "/usr/lib/python3.10/site-packages/pyftdi/ftdi.py", line 384, in create_from_url
    device.open_from_url(url)
  File "/usr/lib/python3.10/site-packages/pyftdi/ftdi.py", line 519, in open_from_url
    self.open_from_device(device, interface)
  File "/usr/lib/python3.10/site-packages/pyftdi/ftdi.py", line 572, in open_from_device
    self._set_interface(config, interface)
  File "/usr/lib/python3.10/site-packages/pyftdi/ftdi.py", line 2027, in _set_interface
    self._in_ep, self._out_ep = endpoints[:2]
ValueError: not enough values to unpack (expected 2, got 1)
```

eblot, 2022-01-15:
Ok, having a look at the actual error, this means that the expected endpoint are not the ones we usually have from a FTDI device... 

Are you really sure this device is a FTDI *232* device?

Please send the output of `lsusb -v -d 0416:5011`

friedrichromstedt, 2022-01-15:
> Are you really sure this device is a FTDI 232 device?

No.  It is expected to be a USB device to control a `RND 320-KA3005P` power supply (`P` stands for *programmable*).  There is some mini language to send commands and retrieve responses.  Might be that I am wrong in supposing that I can use pyftdi to communicate with this equipment.  I believed it to be some non-genuine (non-*FTDI*) chip with similar functionality.  It didn't come with any proprietary control program (only the settings for the *COM port* were given); I barely could find documentation of the control language.
```
Command: lsusb -v -d 0416:5011

Bus 001 Device 010: ID 0416:5011 Winbond Electronics Corp. Virtual Com Port
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            2 Communications
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0416 Winbond Electronics Corp.
  idProduct          0x5011 Virtual Com Port
  bcdDevice            3.00
  iManufacturer           1 Nuvoton
  iProduct                2 USB Virtual COM
  iSerial                 3 002912C90241
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0043
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      1 AT-commands (v.25ter)
      iInterface              0 
      CDC Header:
        bcdCDC               1.10
      CDC Call Management:
        bmCapabilities       0x00
        bDataInterface          1
      CDC ACM:
        bmCapabilities       0x00
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass        10 CDC Data
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0001
  Self Powered
```

eblot, 2022-01-15:
It seems to be a CDC-ACM compliant virtual port, which means that you do not need FTDI at all. This device does really not compatible w/ a FTDI device, which use a proprietary protocol (CDC-ACM is ~ vendor-independent)

Why would you need pyftdi for in this case? If you want to use a serial port w / Python, simply use pyserial with this device.

friedrichromstedt, 2022-01-16:
OK, good to know, I'll dive a little into `pyserial` and continue on the topic next Friday (before I won't have access to the hardware).

There's no other reason for my attempt to use `pyftdi` than lack of knowledge.

friedrichromstedt, 2022-01-21:
Today I successfully communicated with the device using `pyserial`.  The device is accessible as `/dev/ttyACM0`.  I used `$ python3 -m serial.tools.list_ports -v` and `$ python3 -m serial.tools.miniterm /dev/ttyACM0 --echo`.  The former one finds the device as soon as it is attached, but to send and receive data I needed to put another `udev` rule into action:
```
SUBSYSTEMS=="usb", ATTRS{idVendor}=="0416", ATTRS{idProduct}=="5011", ATTRS{serial}=="002912C90241", MODE="0666"
```
It was necessary to spell it `SUBSYSTEMS=="usb"`, not `SUBSYSTEM=="usb"` (the former one features an additional `S`). Using the latter convention (without the additional `S`) prevented the rule from applying successfully.   The `udev` rule I already had in operation:
```
SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", ATTRS{idVendor}=="0416", ATTRS{idProduct}=="0511", TAG+="uaccess"
```
was, curiously, sufficient to make the application of `pyftdi` possible (but failing in the way described earlier).  This rule was based on the rules installed by the Arch Linux `libftdi` package.  The possibility of using `SUBSYSTEMS=="usb"` came to my attention by looking at the output of `$ udevadm info --name /dev/ttyACM0 --attribute-walk`, where it is reported in this way.  During testing it was sufficient to issue `# udevadm control -R`; running `# udevadm trigger` wasn't a necessity (I re-plugged the device).  Temporarily I played with `ATTR{...}` instead of `ATTRS{...}`, but in the end I used `ATTRS`.  Given the observation w.r.t. `SUBSYSTEMS` this might be a good choice.  I used the `ATTR` syntax prior to switching to `SUBSYSTEMS`, at a time the rule apparently didn't apply at all.

There is another, less flexible way of getting the permissions for `/dev/ttyACM0` "right" (i.e., `0666`), by using a rule like:
```
KERNEL=="ttyACM0", MODE="0666"
```

I'd like to be able to control more than one device (a few power supplies aside of a real FTDI component).  Unfortunately, it isn't possible to use `pyserial` with symlinks like that created by adding to the `udev` rule in question:
```
SYMLINK+="ka3005p-01"
```
A rationale is given in https://github.com/pyserial/pyserial/issues/153.  Using `NAME=(...)` in the `udev` rule is possible only for network devices.  I will need to iterate the ports detected by `pyserial` manually, using VID/PID/serial metadata.  In my case, running on Arch Linux, this metadata is properly available (`$ python3 -m serial.tools.list_ports -v`).

friedrichromstedt, 2022-01-28:
The `serial.tools.list_ports` tool has an option `-s` to *list also symlinks* to serial devices.

To test this, I've put the `SYMLINK+="ka3005p-01"` clause in the `udev` rule into operation again. The complete rule is now:
```
SUBSYSTEMS=="usb" ATTRS{idVendor}=="0416", ATTRS{idProduct}=="5011", ATTRS{serial}=="002912C90241", MODE="0666", SYMLINK+="ka3005p-01"
```
The symlink created by this rule now points to `ttyACM0`. Last week it pointed to `bus/usb/001/NNN` with a number `NNN` varying with the unplug/plug iteration.

`$ python3 -m serial.tools.list_ports -s` successfully lists the symlink device:
```
Command: python3 -m serial.tools.list_ports -s

2 ports found
/dev/ka3005p-01     
/dev/ttyACM0        
```
and using the Miniterm on it by `$ python3 -m serial.tools.miniterm /dev/ka2005p-01 --echo` works as well:
```
Command: python3 -m serial.tools.miniterm /dev/ka3005p-01 --echo

--- Miniterm on /dev/ka3005p-01  9600,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
*IDN?
RND 320-KA3005P V4.2 SN:00000331

--- exit ---
```
Last week this command failed:
```
Command: python3 -m serial.tools.miniterm /dev/ka3005p-01 --echo

could not open port '/dev/ka3005p-01': Could not configure port: (25, 'Inappropriate ioctl for device')
```
With this change of situation I might be able to *directly open the symlink device* instead of iterating all ports detected looking for the right device.

I know that all this is quite a bit off-topic for `pyftdi`, but I keep posting to here in order to maintain context.

eblot, 2022-01-28:
Yes, I think it does not relate -at all- with PyFtdi :P

friedrichromstedt, 2022-01-28:
So you'd appreciate it if I'd move the content to somewhere else?

eblot, 2022-01-28:
It is not an issue for me, but I think this info is gonna be quite hard to find. It would be better tied somehow to pyserial.

friedrichromstedt, 2022-01-28:
I just hope that Google fixes the findability problem. Since this isn't an Issue with `pyserial` but rather a documentation of my personal attempts it would probabaly fit best to some personal place of mine.

I'll use the TCP-serial bridge program `tcp_serial_redirect.py` part of the `pyserial` examples. So it looks to me as if the USB/serial communication layer is quite sorted; this issue here is thereby almost completetly resolved.

eblot, 2022-02-13:
Closing as not related to PyFtdi

closed 2022-02-13 by eblot