eblot/pyftdi #288: How slow is I2C with an FT232H?

KaneTW, 2022-02-03:
I'm running firmware_update() at https://gist.github.com/KaneTW/2e2c783bd11fd822dcd677195dabd573 and it's glacially slow, around 70bytes/sec effective. That seems really slow even with the readback that's going on. 

All writes are 8 bytes transmitted, all reads are 4 bytes transmitted and received.
Per 32-byte chunk that's written and readback, we're looking at:
* 12 writes to write and execute
* 1 write/read  poll until it's complete
* 4 writes to execute a readback
* 1 write/read  poll until it's complete
* 8 write/reads to read it back

So about 24 writes and 10 reads best case, or 272 bytes. At 200khz I2C speed, that should be 11ms for a chunk. Real times are closer to 450ms/chunk. Is that normal with the FT232H? Or is something wrong?

eblot, 2022-03-14:
There is one USB round trip for each written byte sequence because of the ACK that needs to be read by the host each time.
I2C is really slow w/ FTDI device, especially with the lack of proper support of the protocol in the device itself.

That said, it really seems to be slow, I have checked the figures since a long time w/ a basic I2C EEPROM device to check that no regression has been added. I have unfortunately no spare time to work on this at the moment. Have you checked w/ a digital scope or a static analyser where the time is actually spent?

P-Myland, 2023-03-09:
I am running into a similar issue. 
I checked the trace:

[image]

The delay between 2 write_to executions (3 bytes: device adress, register, data byte) is approx 16ms. 
This quickly adds quite a heft to any program.
I am looking into minimizing my required writes, but is there something to be done about this huge dead time between each write aswell?

lofuxz, 2023-05-11:
I tried to set the latency to 5ms, which default is 16ms, it works:
self.dev = ftdi.Ftdi()
self.dev.set_latency_timer(5)
you can find the detail of this function in pyftdi.py