eblot/pyftdi #291: CH522 compatibility

charkster, 2022-02-08:
I am seeing my CH552 device detected with `Ftdi.show_devices()`
```
Available interfaces:
  ftdi://ftdi:2232:9D5245F5FA/1   (Sipeed-Debug)
  ftdi://ftdi:2232:9D5245F5FA/2   (Sipeed-Debug)
```
I get errors when I try to use the UART:
```
>>> port = pyftdi.serialext.serial_for_url('ftdi://ftdi:2232:9D5245F5FA/2', baudrate=2000000, timeout=1)
>>> port.write(b'Hello World')
11
>>> port.write(b'Hello World')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/.local/lib/python3.9/site-packages/pyftdi/serialext/protocol_ftdi.py", line 74, in write
    return self.udev.write_data(data)
  File "/home/pi/.local/lib/python3.9/site-packages/pyftdi/ftdi.py", line 1787, in write_data
    length = self._write(data[offset:offset+write_size])
  File "/home/pi/.local/lib/python3.9/site-packages/pyftdi/ftdi.py", line 2098, in _write
    raise FtdiError('UsbError: %s' % str(ex)) from None
pyftdi.ftdi.FtdiError: UsbError: [Errno 110] Operation timed out
```
The CH552 datasheet has a list of all the register addresses, but I could not find a similar list of registers for the FT2232H (to make sure that they are the same).
Here is the link to the CH552 datasheet:
http://www.wch-ic.com/products/CH552.html

Should I not expect compatibility?

charkster, 2022-02-08:
Sorry, I just realized that the CH552 is running a firmware to mimic the FT2232 (with a limitation of 2.5Mbps). As this is a firmware clone of a FT2232H, I will close this issue.

closed 2022-02-08 by charkster

eblot, 2022-02-08:
I'm surprised pyftdi list this device as a FT2232. The only possible way to do so is that the CH552 is configured with the same vendor id as FTDI and the same product id as FT2232. Very, very weird. Are you sure there is no other real FT2232 device connected to your host, for example a USB-JTAG key. The Sipeed-Debug seems to be the key here: it is likely a debugger probe connected to your host on one side and to the JTAG port of the CH552 on the other side.

CH552 and FT2232 are fully unrelated, so you cannot expect any kind of compatibility no. They are fully unrelated.

What you need is a compatible debugger bridge, such as [OpenOCD](https://openocd.org) and GDB although I have no idea whether it supports CH552/8051 devices.

charkster, 2022-02-09:
@eblot Thank you for the advise, and thanks for pyftdi!