eblot/pyftdi #292: How to add USB-2-I2C FT260 to pyftdi

Nippey, 2022-02-09:
Hi,

I just tried to use the pyftdi module with the new FT260Q.
This device enumerates as HID device and provides two channels, one UART and one I2C channel.

I tried to add its VID into ftdi.py, but with no success, I don't find any modules connected.

Q1:
From technical perspective, is pyftdi able to support such device with fixed I2C function?
Q2:
What parts are to be modified for support of this new device?
As of now, I added it to FTDI_VENDOR, DEVICE_NAMES, FIFO_SIZES (dummy entry), FtdiEeprom._PROPERTIES (dummy entry).
But no success so far.

Thanks
Nippey

```
[728.982256] usb 1-1.4: new full-speed USB device number 4 using dwc_otg
[728.130407] usb 1-1.4: New USB device found, idVendor=0403, idProduct=6030, bcdDevice=22.00
[728.130425] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[728.130433] usb 1-1.4: Product: FT260
[728.130441] usb 1-1.4: Manufacturer: FTDI
[728.140622] hid-generic 0003:0403:6030.0001: hidraw0: USB HID v1.11 Device [FTDI FT260] on usb-3f980000.usb-1.4/input0
[728.150873] hid-generic 0003:0403:6030.0002: hidraw1: USB HID v1.11 Device [FTDI FT260] on usb-3f980000.usb-1.4/input1
```

eblot, 2022-02-09:
FT260Q is not a FT*232* device. It uses a fully different protocol (I2C over HID) to manage I2C communication, where FT*232* devices are using FTDI proprietary MPSSE protocol. You cannot use PyFtdi w/ this device, at least for I2C.

There is only little added value to use PyFdti for a pure UART bridge, as all OSes already support the VCP mode out of the box.

Nippey, 2022-02-09:
Thank you for the quick reply, I understand now.
I was already using pyftdi, but due to chip shortage, I switched over to FT260 and wanted to keep using my code.

But then, I'll dive into /dev/hidraw0 ;)

closed 2022-02-09 by Nippey