eblot/pyftdi #295: Wrong setting baudrate for FTDI chip FT230X

jara-k, 2022-03-02:
Hello,
I have problem with setting baudrate on FT230X. E.g. when using subdivisor 0.75, it is converted to 0.25.
Using 0.53.3 with FT230X, baudrate=256000 and CBUS (this one is working fine) on Raspberry OS (64bit), however getting ~267000 (measured on oscilloscope).
Seems similar to linux [bug 210351](https://bugzilla.kernel.org/show_bug.cgi?id=210351) - all subdisivors with `bit 2 == 1` are not working, meaning `FRAC_DIV_CODE >= 4`. When I checked the USB communication, the SetBaudRate request has fields:
```
Baud low: 0x0b
Baud mid: 0x80
lIndex: Port A(0x01)
hIndex: Port 0x00
```
Quick check on Windows machine with FTDI driver however says (notice Index values):
```
Baud low: 0x0b
Baud mid: 0x80
lIndex: Port A(0x00)
hIndex: Port 0x01
```

If there is some easy fix, or path to it - the lattimer is set as well and I am not familiar with it yet -, I can test it.

jara-k, 2022-03-02:
I'll check, no problem.

jara-k, 2022-03-02:
Tried in my environment and working fine. Checked with both oscilloscope and in real setup.
Thanks.

eblot, 2022-03-02:
Thanks for you feedback.

jara-k, 2022-03-02:
Checked for all supported divisors between 11 and 12 inclusive and it works great. Thanks.

eblot, 2022-03-13:
Merged into master, thanks

closed 2022-03-13 by eblot