eblot/pyftdi #309: Unable to configure EEPROM on 232R: No EEPROM decoder for device ?

sutyum, 2022-06-01:
Hi,

We are trying to use the `ftconf.py` script for changing the serial number, the manufacturer name and the product name of the FTDI 232R chip mounted on a breakout board.

The script runs without any prints to stdout or exception when run without `-v` flag. However, with the `-v` verbose flag, it includes two prints that I am unable to decipher: `No EEPROM decoder for device ?`.

Chip used: FTDI 232RL USB Serial

```sh
(ftdi-prog) (base) ..rtualenvs/ftdi-prog-DbuBRRXn❯ ./bin/ftconf.py ftdi://ftdi:232/1 -s ABC1290 -m ABC -p XYZv1 -v
No EEPROM decoder for device ?
No EEPROM decoder for device ?
vendor_id: 0x0403
product_id: 0x6001
type: 0x0000
self_powered: False
remote_wakeup: True
power_max: 90
has_serial: True
suspend_pull_down: False
out_isochronous: False
in_isochronous: False
manufacturer: FTDI
product: FT232R USB UART
serial: A50285BI
```

- OS: MacOS/Ubuntu 20.04LTS
- Python 3.10.4

karaketir16, 2023-02-28:
Your chip is probably a clone because serial is "A50285BI"

refer to [link](https://owenduffy.net/blog/?p=13931)

I also have the same chip, did you find any solution to writing EEPROM?

ohhai, 2024-08-07:
I have same error with Waveshare FT232 USB UART Board from official dealer. Should have genuine FTDI, s/n not A50285BI.
Error is caused by empty "type" field in EEPROM: `self._eeprom[0x06:0x08]` in the code contains zeroes.
I see this type field is described in FTDI docs. Is it ok for it to be just 0x0000?