eblot/pyftdi #316: How to get fdti url?

fredy0219, 2022-09-01:
In my application,I have one FTDI for i2c chip. And I need first initialize i2c
```
i2c.configure('ftdi://ftdi:2232h/1')
```
but I don't want to set it manually every time when I unplug/plug the USB, the url change every time 
Is there any method I can get the url from the beginning and do
```
urls = FTDI.geturls()
i2c.configure(url[0])
```

eblot, 2022-09-01:
>  the url change every time

I do not understand this: do you have an example?

fredy0219, 2022-09-02:
this is my script
```
    i2c = I2cController()
    i2c.configure('ftdi://ftdi:232h:1:2c/1')
    print('connected')
```
I can successfully print "connected" at the first time.
When I unplug the USB and reconnected it again, here's what I got
```
Traceback (most recent call last):
  File "fdti_comunication.py", line 23, in <module>
    i2c.configure('ftdi://ftdi:232h:1:2c/1')
  File "/home/user/.local/lib/python3.8/site-packages/pyftdi/i2c.py", line 498, in configure
    frequency = self._ftdi.open_mpsse_from_url(url, **kwargs)
  File "/home/user/.local/lib/python3.8/site-packages/pyftdi/ftdi.py", line 629, in open_mpsse_from_url
    devdesc, interface = self.get_identifiers(url)
  File "/home/user/.local/lib/python3.8/site-packages/pyftdi/ftdi.py", line 395, in get_identifiers
    return UsbTools.parse_url(url,
  File "/home/user/.local/lib/python3.8/site-packages/pyftdi/usbtools.py", line 328, in parse_url
    raise UsbToolsError('No USB device matches URL %s' %
pyftdi.usbtools.UsbToolsError: No USB device matches URL ftdi://ftdi:232h:1:2c/1
```
somehow the url need to revise to `ftdi://ftdi:232h:1:2d/1`, 2d = 2c+1, then successfully print `connected` 

> > the url change every time

sifive-eblot, 2022-09-06:
It is a bit weird that the device's address changes each time, but you only need NOT to specify the bus:address part:

`ftdi://ftdi:2232h/1` or even simpler if you only use a single FTDI device on your host: `ftdi:///1`

stevessp, 2022-11-08:
Same here. They increment every time I unplug & plug it back in, e.g. ftdi://ftdi:4232:3:e/1 to ftdi://ftdi:4232:3:f/1 to ftdi://ftdi:4232:3:10/1 etc. The address increments every time. Something like urls = FTDI.geturls() would be great.

eblot, 2022-11-08:
I'm sorry but I really do not understand the problem/what you are trying to do/what is not working:

1. If the device has a serial number, use it as a unique identifier, and do not use the bus:address notation
2. If your device is unique on your host (single FTDI device of one kind), you do not need to use the bus:address notation
3. If your device is not unique on your host (multiple FTDI devices of the same kind) - and your host is unable to report the same bus:address when you plug the device back, do not use the bus:address notation and program a serial number, then use it (solution 1). If you cannot program a serial number, then you are out of luck, PyFTDI cannot uniquely identified a device if there is no unique information available.

But again, I do not understand what is the real issue you are trying to circumvent...

stevessp, 2023-01-24:
Now that I understand the problem better (not a hardware engineer)...

Unfortunately, there are multiple devices, they don't have serial numbers, and they're on a customer's board so we can't use solutions 1 or 2. However, I discovered your list_devices() function so what I've been doing is activating them one at a time. That way I can identify each as it comes up and can track it in code. Works well enough as long as I stay in the same session.

It is weird that they get different addresses each time, but I think that's just how USB works? I've found it's the same on Windows or Mac. Thanks for your work.

RomainWCSBx, 2024-03-28:
Got the same issue. Here is my home-made solution:

```python
from pyftdi.usbtools import UsbTools
from pyftdi.ftdi import Ftdi

def get_ftdi_devices_url():
    f=Ftdi()
    return UsbTools.build_dev_strings(scheme="ftdi", vdict=f.VENDOR_IDS, pdict=f.PRODUCT_IDS, devdescs=Ftdi.list_devices())
```