eblot/pyftdi #322: Rasberry Pi 4

Notselwyn, 2022-10-01:
Is possible to use `pyftdi` on a Raspberry Pi 4 with its' GPIO acting as SPI? I intend to use it for [pyspiflash](https://github.com/eblot/pyspiflash).

I tried looking at the documentary of `pyftdi` but it didn't make me any wiser.

enkiusz, 2022-10-27:
You can look at my simple script that uses the GPIO (called CBUS in FTDI) feature to drive some outputs: https://github.com/enkiusz/tools/blob/master/bin/state-actuator.py. When you ignore the mqtt stuff it boils down to the below code:

```
CBUS3 = 0b1000
CBUS2 = 0b0100

        ftdi = Ftdi()
        ftdi.open_from_url(url=config.url)

        # Set bitbang mode
        ftdi.set_bitmode(0, Ftdi.BitMode.CBUS)

        # Set output mode on both CBUS3 and CBUS2 pins
        ftdi.set_cbus_direction(CBUS3|CBUS2, CBUS3|CBUS2)

        # Set initial state
        set_state(ftdi, 0)

        # Drive CBUS3 and CBUS2 high
        ftdi.set_cbus_gpio(CBUS3 | CBUS2)
```
You can get the FTDI device URL by running the [ftdi_urls.py](https://github.com/eblot/pyftdi/blob/master/pyftdi/bin/ftdi_urls.py) script. The ftdi_urls.py should detect the device and show the URL for it, like below:

```
davinci:~# python3 ftdi_urls.py
Available interfaces:
  ftdi://ftdi:232:FTGS6NLO/1   (USB Serial Converter)
```

eblot, 2023-01-28:
PyFTDI is solely for FTDI devices. So as long as you are not using an FTDI device connected to your RPi, the answer is no.

closed 2023-01-28 by eblot