eblot/pyftdi #332: pyftdi not acquiring serial num/id from USB dongle on Ubuntu

insane-dreamer, 2023-01-27:
We have a FTDI USB dongle for a biosignal device that registers with the OS USB kernel, and shows up when listing devices with pyftdi, but doesn't return the serial num or any other info. This is on Ubuntu 21.10 with the latest pyftdi release. 

Here's what the syslog shows when plogging in the device: 
[image]

But when querying with pyftdi: 
[image]

The same FTDI dongle works with the FDTI official DLLs on Windows. However, using the FTDI .so for Linux + the ftd2xx python library yields the same results as with pyftdi, so whatever the issue is, neither library is picking up the dongle.

eblot, 2023-01-28:
Have you checked [this error message](https://eblot.github.io/pyftdi/troubleshooting.html#error-the-device-has-no-langid) against the troubleshooting guide?

insane-dreamer, 2023-02-02:
That resolved it. Thanks!

closed 2023-02-02 by insane-dreamer