eblot/pyftdi #342: Intermittent "the device has no langid" error, can't be cleared without a new process

nmichaels-qualinx, 2023-05-03:
I have a device (0403:6010 Future Technology Devices International, Ltd FT2232C/D/H Dual UART/FIFO IC) that I've been using pyftdi with successfully for a while now. However, occasionally when I call `Ftdi.list_devices()` I get a `ValueError` telling me the device has no langid.

I managed to get this to happen and drop into an interactive shell. Repeated calls to `Ftdi.list_devices()` still raise the same exception. Reloading pyftdi didn't change it. However, if I run a script that calls `list_devices()` in a subprocess or through `os.execv`, it works.

So it looks like something tied to the process is preventing this error from clearing. I'm running this on a VM (Debian on proxmox) that has the port passed through to it. I'm also disabling and re-enabling the port with a [USB switch](https://www.yepkit.com/product/300115/YKUSHXS). Here's the most reliable way I've found to reproduce the issue:

```python
#!/usr/bin/env python3
# Call this test.py
from pyftdi.ftdi import Ftdi

try:
    Ftdi.list_devices()
except Exception as e:
    import code
    code.interact(local={'e':e, 'Ftdi':Ftdi})
    import sys
    sys.exit(-1)
print('.', end='')
```
`$ while true ; do ./test.py || break ; done`

While that's running, trigger a test, which basically unplugs then replugs the device I care about.

[This comment](https://github.com/pyusb/pyusb/issues/139#issuecomment-565114069) seems to say that there's a potential race condition where if the device isn't immediately available after being configured it might return this error. That seems like a plausible explanation.

Once this happens, the process is effectively useless. Is there a way to reinitialize it that I'm not seeing?

eblot, 2023-05-03:
I'm not sure whether it could help, but have you tried to [flush the device cache](https://eblot.github.io/pyftdi/api/usbtools.html#pyftdi.usbtools.UsbTools.flush_cache)?

If it is a PyUSB issue, I'm not sure what can be done @ PyFtdi level but ideas are welcomed.

nmichaels-qualinx, 2023-05-03:
> I'm not sure whether it could help, but have you tried to [flush the device cache](https://eblot.github.io/pyftdi/api/usbtools.html#pyftdi.usbtools.UsbTools.flush_cache)?
> 
> If it is a PyUSB issue, I'm not sure what can be done @ PyFtdi level but ideas are welcomed.

I had not, and it worked. Thanks!

closed 2023-05-03 by nmichaels-qualinx