eblot/pyftdi #344: LSB first with MPSSE SPI?

wuzhongjie1997, 2023-06-08:
First of all, thanks for this wonderful project, it allows me to convert an FT2232H dongle into a swiss army knife in embedded debugging!

My question is as follows:

In an STM32 MCU, SPI ports can operate either MSB first or LSB first. No such option seems to exist in the SPI API docs, but according to FTDI [AN114](https://www.ftdichip.com/Support/Documents/AppNotes/AN_114_FTDI_Hi_Speed_USB_To_SPI_Example.pdf) Page 3, MPSSE supports both operating modes. 

As confirmed using a logic analyzer, when i follow the SPI API [example](https://eblot.github.io/pyftdi/api/spi.html?highlight=lsb%20first) the data is sent MSB first. So, other than manually converting each byte in the input (and output) array, how can I configure MPSSE SPI to transmit/receive LSB first?

Thanks for reviewing!

eblot, 2023-06-08:
> First of all, thanks for this wonderful project, it allows me to convert an FT2232H dongle into a swiss army knife in embedded debugging!

Thanks.

> So, other than manually converting each byte in the input (and output) array, how can I configure MPSSE SPI to transmit/receive LSB first?

Up to now this feature has never been asked and is not supported (as it would require just another combination of options, tests, etc.) but it should be moderately trivial to implement this.

You would need to replace all `Ftdi.*_MSB` constants in `_exchange_half_duplex` and `_exchange_full_duplex` methods from `spi.py` file with their `Ftdi.*_LSB` counterparts, see `# MPSSE Commands` around line 148 in `ftdi.py` file.

HTH.

wuzhongjie1997, 2023-06-12:
Thank you for the reply, closing as completed.

closed 2023-06-12 by wuzhongjie1997