eblot/pyftdi #346: Retry issue with SPI reads

andreasWallner, 2023-06-28:
We are currently experiencing some issues with SPI over a FT4232. `.read()` will sometimes return an empty bytearray, which will then throw off future rx. From our analysis this seems to be caused by the limit of 4 retries in https://github.com/eblot/pyftdi/blob/05a36f9474464157c1e30c6daf00e190044beb1a/pyftdi/spi.py#L860 - increasing that fixes our issue.

In the example below we are sending 32 byte data to an embedded device, that device will then echo those 32 byte back. The loop in the code is due to us wanting to poll until the embedded device sends the first non-zero byte, but this does not happen in the example (since it's not processing anything)
```python 
import logging
from pyftdi import FtdiLogger
from pyftdi.spi import SpiController

logging.basicConfig(level=logging.DEBUG)
logger = logging.getLogger("main")
FtdiLogger.set_level(logging.DEBUG)

ctrl = SpiController()
ctrl.configure("ftdi:///2", debug=True)
spi = ctrl.get_port(cs=0, freq=2000, mode=2)

EXCHANGE_BYTES = bytearray([i for i in range(1, 33)])
i = 0
for _ in range(2):
    spi.write(EXCHANGE_BYTES)

    while True:
        resp = spi.read(1, stop=False)
        logger.debug(f"rx {len(resp)}: " + resp.hex())
        if len(resp) != 0 and resp[0] != 0:
            break

    resp += spi.read(31)

    suc = resp == EXCHANGE_BYTES
    if not suc:
        print(resp.hex())
        print(EXCHANGE_BYTES.hex())
    print()
    i += 1 
```
The above described behavior occurs twice during the first transfer, which will lead to this:
[image]

So we can see that the FTDI has actually read 34 bytes, instead of the 32 that we expected.

<details>
<summary>Snip from the log that shows the problem:</summary>
```
DEBUG:pyftdi.ftdi:> 8b86b70b
DEBUG:pyftdi.ftdi:< 3260
DEBUG:pyftdi.ftdi:rx buf purged
DEBUG:pyftdi.ftdi:Clock frequency: 2.000 KHz (error: +0.0 %)
DEBUG:pyftdi.ftdi:> 800d0b80050b101f000102030405060708090a0b0c0d0e0f101112131415161718191a1b1c1d1e1f2080050b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b80080b
DEBUG:pyftdi.ftdi:> 800d0b80050b20000087
DEBUG:pyftdi.ftdi:< 3200
DEBUG:pyftdi.ftdi:< 3200
DEBUG:pyftdi.ftdi:< 3200
DEBUG:pyftdi.ftdi:< 3200
DEBUG:main:rx 0:
DEBUG:pyftdi.ftdi:> 800d0b80050b20000087
DEBUG:pyftdi.ftdi:< 3200
DEBUG:pyftdi.ftdi:< 3200
DEBUG:pyftdi.ftdi:< 3200
DEBUG:pyftdi.ftdi:< 3200
DEBUG:main:rx 0:
DEBUG:pyftdi.ftdi:> 800d0b80050b20000087
DEBUG:pyftdi.ftdi:< 320001
DEBUG:main:rx 1: 01
DEBUG:pyftdi.ftdi:> 800d0b80050b201e008780050b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b800d0b80080b
DEBUG:pyftdi.ftdi:< 32000203
DEBUG:pyftdi.ftdi:< 3200040506
[... cut ...]
DEBUG:pyftdi.ftdi:< 320013141516
DEBUG:pyftdi.ftdi:< 32001718191a
DEBUG:pyftdi.ftdi:< 32001b1c1d1e
DEBUG:pyftdi.ftdi:< 32001f200000
```

</details>

For a workaround: We tried to increase the retry count to 16 - that fixes our issue. 
Before opening a PR I wanted to check if there was a specific reason to go with 4 there, does increasing that have negative consequences somewhere else? Should an exception be raised in this case, especially considering that future receptions can carry over this "offset"?

eblot, 2023-06-29:
I think I never run FTDI SPI as such as low clock speed, so maybe the timeout should be increased with such low-extreme clock rate, yes. The timeout is there to manage some jittery behavior between the USB time frame, the FTDI latency response, etc. Did you try to change the default latency to see if this improves the overall behavior?

If not, maybe this retry count could be increased for very low clock speed, yes. The main drawback is increased latency, so this could only be done when the clock drops between some low value, that is yet to be found :)

andreasWallner, 2023-06-29:
I'll give it a try - interestingly we don't see this issue with the even slower clock speed of 1kHz.
Wouldn't the latency only be increased in case we don't have a response yet (which should actually never happen in the SPI case)?

andreasWallner, 2023-09-26:
Just a quick sign of live - I haven't forgotten but time has been in short supply lately. I'll ask a student to have a look and evaluate.

haberzdomi, 2023-11-22:
Hi, I am this student of @andreasWallner and have tried different FTDI latencies for the above example.
Indeed, I found that a higher latency makes the example work and for high latencies the clock speed can be reduced to very lower frequencies, although the use case can be questioned of course.
Thanks a lot for helping with this issue.

andreasWallner, 2023-11-24:
@eblot 
From my POV we could close this since we do have a workaround with manually setting the latency.

closed 2024-07-20 by andreasWallner