eblot/pyftdi #347: 0.54.0: Deprecated setuptools functionality in use, will break on 2023-10-30

dvzrv, 2023-07-26:
Hi! When trying to package this project for Arch Linux I ran into an explicit warning about the setuptools setup:

```
* Building wheel...
/usr/lib/python3.11/site-packages/setuptools/_distutils/dist.py:265: UserWarning: Unknown distribution option: 'test_requires'
  warnings.warn(msg)
/usr/lib/python3.11/site-packages/setuptools/config/setupcfg.py:293: _DeprecatedConfig: Deprecated config in `setup.cfg`
!!

        ********************************************************************************
        The license_file parameter is deprecated, use license_files instead.

        By 2023-Oct-30, you need to update your project and remove deprecated calls
        or your builds will no longer be supported.

        See https://setuptools.pypa.io/en/latest/userguide/declarative_config.html for details.
        ********************************************************************************

!!
  parsed = self.parsers.get(option_name, lambda x: x)(value)
```

eblot, 2023-08-05:
Thanks for reporting this issue

eblot, 2023-08-07:
Updated in 0.55. Unfortunately I do not know when it can be made available since PyPI new upload strategy with 2FA is just a gigantic mess and I really have no time to spend reading through their numerous documentation. I guess I should use GitHub Actions as the new way, but I'm really tired of this ....

eblot, 2023-08-12:
Available on [PyPI](https://pypi.org/project/pyftdi/0.55.0/)

closed 2023-08-12 by eblot