eblot/pyftdi #351: 0.54.0 FT230x EEPROM API inaccuracies

timothy-lee2415, 2023-08-09:
Using pyftdi to bitbang on the FT230x CBUS pins. I want to check if CBUS functions are set to GPIO, and if not, set them to GPIO in EEPROM so I can use the cbus gpio APIs. I ran into a number of issues along with some inaccuracies with the APIs:
Doing:
`    ftdi = Ftdi()
    ftdi.open_from_url("ftdi://ftdi:ft-x:/1")
    if ftdi.has_cbus:
        print("Found CBUS enabled FTDI 230x")

    eeprom = FtdiEeprom()
    eeprom.connect(ftdi)
    eeprom.set_property("cbus_func_0", "GPIO")
    eeprom.set_property("cbus_func_1", "GPIO")
    eeprom.set_property("cbus_func_2", "GPIO")
    eeprom.set_property("cbus_func_3", "GPIO")
    print(eeprom.commit(dry_run=False))
    eeprom.reset_device()
    eeprom.dump_config()`

- eeprom.commit always returns false even if one of the above options is changed. e.g. if I write DRIVE0 with eeprom.set_property and commit,  eeprom.commit still returns False. This is an issue since I need to be able to tell if settings were changed or not so I can avoid doing a lengthy reset. Some kind of get_property API would be incredibly useful here.
- eeprom.sync() doesn't seem to apply the EEPROM changes, only after a device reset do the settings take place.

eblot, 2023-08-12:
Yeap, EEPROM support is mostly broken. Patches accepted.

JonathonReinhart, 2023-08-18:
I also just noticed that `Eeprom.commit()` always returns the value of the `dry_run` argument:

https://github.com/eblot/pyftdi/blob/0d20ccd1ae4542d868a252c492ba6855b4e67906/pyftdi/eeprom.py#L657

Assuming `self._ftdi.overwrite_eeprom(self._eeprom, dry_run=dry_run)` does the right thing, it seems this should be:

```python
        return not dry_run
```