eblot/pyftdi #352: 0.55.0 not on pypi.org

dvzrv, 2023-08-10:
Hi! I package this project for Arch Linux.

For upstream tracking purposes I rely on the versions available on pypi.org. Unfortunately the latest version 0.55.0 is not available there, but a tag has been issued in this repository.

eblot, 2023-08-10:
Yes, see my comment here https://github.com/eblot/pyftdi/issues/347#issuecomment-1667236048

I'm f... tired of the 2FA/MFA user-unfriendly procedures, with each site deploying its own system, popping out dozens of solutions and workarounds. So till I find a ready-to-use script from PyPI site or a user provides a PyFtdi patch to ease this process, I'm afraid 0.55+ won't make it to PyPI.

Sorry about that, but I have no spare time (and no real will to be honest) to dig into this work, but patches are always welcomed. PyFtdi already runs GitHub Actions so I believe it could leverage them to upload versions on PyPI. I just do not want to lose any more time with PyPI security issues and lack of easy-to-use solutions. Maybe I did not find the proper, simple documentation though.

dvzrv, 2023-08-10:
Ah... sorry, interacting with too many projects makes me forget who wrote what :sweat_smile: 

Irt deploying to pypi.org, it seems that should still be somewhat easy.
For repod, I do the following to deploy (this is using pdm as project management tooling though):
https://gitlab.archlinux.org/archlinux/repod/-/blob/122ac2a638bb2104a4447d162a13be0b06bd1b02/.gitlab-ci.yml#L170-183

In your current setup, are you using an auth token?
Either way, here is the documentation that should be able to help you to upload to pypi.org: https://packaging.python.org/en/latest/tutorials/packaging-projects/#uploading-the-distribution-archives
A little bit further up is also further info on various project management tooling.

eblot, 2023-08-12:
> Ah... sorry, interacting with too many projects makes me forget who wrote what 😅
No problem.
 
I've created an auth token. Till PyPI decide it is not secure enough and enforce their trusted publisher scheme...
0.55 should be available on PyPI.

closed 2023-08-12 by eblot