eblot/pyftdi #358: Intermittent errors when accessing FTDI

srobertjames, 2023-09-20:
I've setup pyftdi on Ubuntu, including the udev rules.  Sometimes the `ftdi_urls` script can see my FTDI, sometimes it can't.  This happens without any changes to perms.

Consistently:
1. The first read works
2. Subsequent reads fail with that error
3.  Until I wait a while (perhaps 100 seconds?) - then the _next_ read works (but not more)

```
etc/udev/rules.d$ ftdi_urls.py 

Error: The device has no langid (permission issue, no string descriptors supported or device error)
/etc/udev/rules.d$ ftdi_urls.py 
Available interfaces:
  ftdi://ftdi:232:FT9PIFW4/1   (TTL232R-3V3)

/etc/udev/rules.d$ ftdi_urls.py 

Error: The device has no langid (permission issue, no string descriptors supported or device error)
```

My udev file is:
```
/etc/udev/rules.d$ cat 11-ftdi.rules 
# /etc/udev/rules.d/11-ftdi.rules

# FT232AM/FT232BM/FT232R
SUBSYSTEM=="usb", ATTR{idVendor}=="0403", ATTR{idProduct}=="6001", GROUP="dialout", MODE="0664"
# FT2232C/FT2232D/FT2232H
SUBSYSTEM=="usb", ATTR{idVendor}=="0403", ATTR{idProduct}=="6010", GROUP="dialout", MODE="0664"
# FT4232/FT4232H
SUBSYSTEM=="usb", ATTR{idVendor}=="0403", ATTR{idProduct}=="6011", GROUP="dialout", MODE="0664"
# FT232H
SUBSYSTEM=="usb", ATTR{idVendor}=="0403", ATTR{idProduct}=="6014", GROUP="dialout", MODE="0664"
# FT230X/FT231X/FT234X
SUBSYSTEM=="usb", ATTR{idVendor}=="0403", ATTR{idProduct}=="6015", GROUP="dialout", MODE="0664"
# FT4232HA
SUBSYSTEM=="usb", ATTR{idVendor}=="0403", ATTR{idProduct}=="6048", GROUP="dialout", MODE="0664"
```

My `lsusb -v`  
```
Bus 001 Device 011: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x0403 Future Technology Devices International, Ltd
  idProduct          0x6001 FT232 Serial (UART) IC
  bcdDevice            6.00
  iManufacturer           1 FTDI
  iProduct                2 TTL232R-3V3
  iSerial                 3 FT9PIFW4
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0020
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower               90mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              2 TTL232R-3V3
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0000
  (Bus Powered)
```

There's also:
```
crw-rw---- 1 root dialout 188, 0 Sep 20 13:35 /dev/ttyUSB0
```

I am a member of the dialout group.

eblot, 2023-09-20:
Questions:
1. Is it a genuine FTDI or one of those crap chinese fake FTDIs?
2. When PyFtdi returns an error, what does return `lsusb -v` if you run it _immediately_ after the issue
3. Are there some related error message reported by the kernel (see `dmesg`)

srobertjames, 2023-09-20:
Questions:

> Is it a genuine FTDI or one of those crap chinese fake FTDIs?

I certainly _wanted_ a genuine FTDI, but it's hard to tell.  The bag has the FTDI logo on it, and says "FTDI Chip".  How can I verify if it's genuine?

> When PyFtdi returns an error, what does return lsusb -v if you run it immediately after the issue

(will do)

> Are there some related error message reported by the kernel (see dmesg)

```
/etc/udev/rules.d$ sudo dmesg | grep -i -2 -E '(FTDI)|(Future)|(error)' 
[    3.402108] evm: HMAC attrs: 0x1
[    3.402866] PM:   Magic number: 7:134:136
[    3.404605] RAS: Correctable Errors collector initialized.
[    3.411067] Freeing unused decrypted memory: 2036K
[    3.412459] Freeing unused kernel image (initmem) memory: 4764K
--
[164606.603377] usb 1-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[164606.603383] usb 1-2.1: Product: TTL232R-3V3
[164606.603388] usb 1-2.1: Manufacturer: FTDI
[164606.603392] usb 1-2.1: SerialNumber: FT9PIFW4
[164606.698664] usbcore: registered new interface driver usbserial_generic
[164606.698683] usbserial: USB Serial support registered for generic
[164606.707692] usbcore: registered new interface driver ftdi_sio
[164606.707717] usbserial: USB Serial support registered for FTDI USB Serial Device
[164606.712189] ftdi_sio 1-2.1:1.0: FTDI USB Serial Device converter detected
[164606.713544] usb 1-2.1: Detected FT232R
[164606.721419] usb 1-2.1: FTDI USB Serial Device converter now attached to ttyUSB0
[165341.725472] usb 1-2.1: USB disconnect, device number 9
[165341.732718] ftdi_sio ttyUSB0: FTDI USB Serial Device converter now disconnected from ttyUSB0
[165341.732783] ftdi_sio 1-2.1:1.0: device disconnected
[165380.077401] usb 1-2.1: new full-speed USB device number 10 using uhci_hcd
[165380.437881] usb 1-2.1: New USB device found, idVendor=0403, idProduct=6001, bcdDevice= 6.00
[165380.437904] usb 1-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[165380.437914] usb 1-2.1: Product: TTL232R-3V3
[165380.437921] usb 1-2.1: Manufacturer: FTDI
[165380.437929] usb 1-2.1: SerialNumber: FT9PIFW4
[165380.475561] ftdi_sio 1-2.1:1.0: FTDI USB Serial Device converter detected
[165380.475795] usb 1-2.1: Detected FT232R
[165380.481244] usb 1-2.1: FTDI USB Serial Device converter now attached to ttyUSB0
[165981.453687] usb 1-2.1: USB disconnect, device number 10
[165981.458696] ftdi_sio ttyUSB0: FTDI USB Serial Device converter now disconnected from ttyUSB0
[165981.458754] ftdi_sio 1-2.1:1.0: device disconnected
[165981.902250] usb 1-2.1: new full-speed USB device number 11 using uhci_hcd
[165982.299340] usb 1-2.1: New USB device found, idVendor=0403, idProduct=6001, bcdDevice= 6.00
[165982.299348] usb 1-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[165982.299351] usb 1-2.1: Product: TTL232R-3V3
[165982.299354] usb 1-2.1: Manufacturer: FTDI
[165982.299356] usb 1-2.1: SerialNumber: FT9PIFW4
[165982.341317] ftdi_sio 1-2.1:1.0: FTDI USB Serial Device converter detected
[165982.341359] usb 1-2.1: Detected FT232R
[165982.349832] usb 1-2.1: FTDI USB Serial Device converter now attached to ttyUSB0
[166920.664039] usb 1-2.1: USB disconnect, device number 11
[166920.664771] ftdi_sio ttyUSB0: FTDI USB Serial Device converter now disconnected from ttyUSB0
[166920.664911] ftdi_sio 1-2.1:1.0: device disconnected
[166925.141807] usb 1-2.1: new full-speed USB device number 12 using uhci_hcd
[166925.505150] usb 1-2.1: New USB device found, idVendor=0403, idProduct=6001, bcdDevice= 6.00
[166925.505170] usb 1-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[166925.505176] usb 1-2.1: Product: TTL232R-3V3
[166925.505181] usb 1-2.1: Manufacturer: FTDI
[166925.505185] usb 1-2.1: SerialNumber: FT9PIFW4
[166925.552294] ftdi_sio 1-2.1:1.0: FTDI USB Serial Device converter detected
[166925.552389] usb 1-2.1: Detected FT232R
[166925.559078] usb 1-2.1: FTDI USB Serial Device converter now attached to ttyUSB0
[167299.871095] usb 1-2.1: usbfs: interface 0 claimed by ftdi_sio while 'ipython' sets config #1
[167299.872134] ftdi_sio ttyUSB0: FTDI USB Serial Device converter now disconnected from ttyUSB0
[167299.872170] ftdi_sio 1-2.1:1.0: device disconnected
```

I'll add that I get similar behavior with other ops, e.g. if I try to open a port `USBTimeoutError: [Errno 110] Operation timed out`.

srobertjames, 2023-09-20:
I have a clue as to the problem.  See https://unix.stackexchange.com/questions/696001/dev-ttyusb0-is-available-but-after-try-to-call-its-gone which posted a similar problem, which was due to `usb 1-2: usbfs: interface 0 claimed by ftdi_sio while 'brltty' sets config #1` from `brltty` daemon.  I don't have `brltty` and it does not appear anywhere in `dmesg`.

However, look at this (very similar) line: `usb 1-2.1: usbfs: interface 0 claimed by ftdi_sio while 'ipython' sets config #1`.  The ipython process is of course what I was using to use `pyftdi`.

I'm not sure how to take this further, or what that error indicates.

eblot, 2023-09-21:
I would suggest to first try with a simple(r) environment, _i.e._ `cpython`

srobertjames, 2023-09-21:
Note that I got similar behavior when just using `ftdi_urls` from the command line.

For future reference, here's an (old) post talking about similar issues in libftdi itself, having to do with attach and reattach: https://libftdi.developer.intra2net.narkive.com/4QoVetCi/disconnect-by-usbfs-dev-ttyusb0-disappears-when-using