eblot/pyftdi #372: in_waiting and out_waiting in pyftdi/serialext/protocol_ftdi.py not implemented but return always 0

SeDi343, 2024-03-25:
I was debugging more than half a day why my readup thread is not working using this properties. Please consider raising an NotImplemented(). To reduce unnecessary waste of time for developers.

pyftdi/serialext/protocol_ftdi.py
```
    @property
    def in_waiting(self):
        """Return the number of characters currently in the input buffer."""
        # not implemented
        return 0

    @property
    def out_waiting(self):
        """Return the number of bytes currently in the output buffer."""
        return 0
```

eblot, 2024-03-25:
If I'm not mistaken (since I have not work on this code for years), `in_waiting` is used in pyserial and other APIs that do not expect an exception unless something not recoverable (such as the communication port being closed) occurs.

The semantic might have changed since the initial development of this file, so maybe it is no longer the case, but if it has not, it would not possible to change the behavior of this function without breaking its purpose unfortunately...

SeDi343, 2024-03-26:
Ended up using serial, since those properties are mandatory for me.