eblot/pyftdi #377: No such device / 4 USB devices match URL 'ftdi:///2'

Shoyur, 2024-04-21:
Hi,

I plugged the FT4232h mini module in Windows 11, i used Zadig to switch the 4 interfaces drivers from FTDI to usblib0.
Now i can use I2C on port 1/4 with pyftdi
i2c = I2cController()
i2c.configure('ftdi://0x0403:0x6011:1/1')
and this works perfectly with a nice network of I2C devices
but any other 3 ports
gpio_encoders = GpioAsyncController()
gpio_encoders.configure('ftdi:///2', direction=0b00011111)
or whatever address i try to use, i always get an Exception with Error
No such device [or] 4 USB devices match URL 'ftdi:///2'
and
Ftdi.show_devices()
prints :
Available interfaces:
  ftdi://ftdi:4232:FT7GOZHF/1   (FT4232H MiniModule)
  ftdi://ftdi:4232:FT7GOZHF/1   (FT4232H MiniModule)
  ftdi://ftdi:4232:FT7GOZHF/1   (FT4232H MiniModule)
  ftdi://ftdi:4232:FT7GOZHF/1   (FT4232H MiniModule)
  
i really dont know why they all have the same address..

Please help me. Thank you.

Shoyur, 2024-04-23:
I wish i could delete this issue, cause I have found the solution.
At first, I have read somewhere that i need to be sure I installed the same driver on all ports, but it misled me cause in fact this file :
_[](https://eblot.github.io/pyftdi/installation.html)_
says :
_With FTDI devices with multiple channels, such as FT2232 (2 channels) and FT4232 (4 channels), you must install the driver for the composite parent, not for the individual interfaces._
So now everything works.
Thank you.

eblot, 2024-04-23:
Sorry not to have answerer sooner.

It is important not to delete such a ticket as other users might bump into the same problem and look for the solution :-) Simply closing it (with the rationale, which you did) is the right way to do. Thanks.