eblot/pyftdi #388: pyftdi for windows

rqx110, 2024-11-06:
there is a fork project provides a better out-of-the-box experience by supporting the default Windows plug-and-play drivers provided by FTDI. https://github.com/mariusgreuel/pyftdiwin

could you merge these codes?

eblot, 2024-11-19:
Yes I guess I could but it would require to create pull request(s). If you know the author or volunteer to create those PRs, please let me know.

eblot, 2024-11-23:
I had a quick look at this project/fork.

From what I understand, I think it would make much more sense to add D2XX support to [pyusb](https://github.com/pyusb/pyusb) rather than to PyFtdi:
1. It could be used for other pyusb-based projects, not only PyFtdi
2. PyFtdi is designed not to have a direct dependencies on a native library. I'm pretty sure having a requirement on `ctypes` would create issues with other projects that rely on PyFtdi. This means this would have to be an optional dependency, which might be not simple to handle.
3. Testing is also an real issue :(

eblot, 2024-11-23:
BTW I've added a direct link to this project from https://eblot.github.io/pyftdi/requirements.html#development

mcuee, 2025-09-10:
I do not think it makes sense for the function to be added in pyusb.

Your best bet is to improve https://github.com/mariusgreuel/pyftdiwin.

I have closed https://github.com/pyusb/pyusb/issues/539 as not planned.

mcuee, 2025-09-10:
BTW, the author of pyftdiwin also did the following.
https://github.com/avrdudes/libftdi

NsTrmb, 2025-09-24:
FYI : on my side i'm using WSL2 with pyftdi + USB binding from Windows to Linux and all is working like a charm 😸

SpareSimian, 2025-12-19:
If someone picks up the other project, I found two forks with small fixes:
https://github.com/mariusgreuel/pyftdiwin/compare/main...jonathanlarochelle:pyftdiwin:master
https://github.com/mariusgreuel/pyftdiwin/compare/main...SamHerts:pyftdiwin_eepromWrite_bugfix:main

The tools in pyftdiwin fail I think because the internal import paths assume the Linux package layout with subdirectory bin and the tools are deployed under the scripts directory, instead. 

My own motivation is to build some small utilities for communicating with a device that uses the D2XX driver and talks Modbus. I figure I'll fork pyftdiwin and merge the two commits above, then fix the tool paths.