eblot/pyftdi #389: Getting-started help (pyftdi on Windows 10 using Python 3.1.13 and single FTDI232H module)

wb0gaz, 2024-11-08:
I am beginning work with pyftdi version 0.55.3 (on Windows 10, under Python 3.13.0, with a single FDT232H module attached). My (small) initial test program listing is below, and the printed text from the progrm follows. There are two concerns - about open_from_url() return value, and ftdiEeprom API.

Some initial steps are working - Ftdi.show_devices() and Ftdi.list_devices(), but I am not sure about result from open_from_url(), and initial attempt to dump EEPROM cannot start.

Question - open_from_url() is returning "none". Is this correct? 
I am following "ftdi - FTDI low-level driver".

Question - what is cause of the error "NameError: name 'ftdiEeprom' is not defined" when I try to inspect data in EEPROM? 
I am following documentation section "eeprom - EEPROM API Quickstart".

I have installed pyftdi and used Zadeg to create the required libusb-win32 device driver interface.

Below is my test program. Following that is the output I have obtained.

print("---------------------------------------------------------")
print("import pyftdi")
import pyftdi
print("---------------------------------------------------------")
print("from pyftdi.ftdi import Ftdi")
from pyftdi.ftdi import Ftdi
print("---------------------------------------------------------")
print ("Ftdi.show_devices()")
print(Ftdi.show_devices())
print("---------------------------------------------------------")
print ("Ftdi.list_devices()")
print(Ftdi.list_devices())
print("---------------------------------------------------------")
print("ftdi - FTDI low-level driver")
print("---------------------------------------------------------")
print ("interface = Ftdi().open_from_url('ftdi:///1')")
interface = Ftdi().open_from_url('ftdi:///1')
print("interface = ", interface)
print("---------------------------------------------------------")
print("eeprom - EEPROM API Quickstart")
print("---------------------------------------------------------")
print ("eeprom = ftdiEeprom()")
eeprom = ftdiEeprom()
print("eeprom = ", eeprom)
print("---------------------------------------------------------")
print("eeprom.open('ftdi://ftdi:2232h/1')")
eeprom.open('ftdi://ftdi:2232h/1')
print("---------------------------------------------------------")
print("eeprom.dump_config()")
eeprom.dump_config()
print("---------------------------------------------------------")

Below is the output I have obtained so far.

---------------------------------------------------------
import pyftdi
---------------------------------------------------------
from pyftdi.ftdi import Ftdi
---------------------------------------------------------
Ftdi.show_devices()
Available interfaces:
  ftdi://ftdi:232h:0:1/1  (Single RS232-HS)

None
---------------------------------------------------------
Ftdi.list_devices()
[(UsbDeviceDescriptor(vid=1027, pid=24596, bus=0, address=1, sn=None, index=None, description='Single RS232-HS'), 1)]
---------------------------------------------------------
ftdi - FTDI low-level driver
---------------------------------------------------------
interface = Ftdi().open_from_url('ftdi:///1')
interface =  None
---------------------------------------------------------
eeprom - EEPROM API Quickstart
---------------------------------------------------------
eeprom = ftdiEeprom()
Traceback (most recent call last):
  File "C:\Users\user\Desktop\pantilt\a.py", line 23, in <module>
    eeprom = ftdiEeprom()
             ^^^^^^^^^^
NameError: name 'ftdiEeprom' is not defined

juanitoetc, 2024-11-09:
> Question - open_from_url() is returning "none". Is this correct?
I am following "ftdi - FTDI low-level driver".

No, that is not correct. I dont use Windows but I assume you are able to communicate with the device because it answers `Ftdi.show_devices()`. Your problem is that you are trying to use a class method when the API says it is a instance method.
Simple replace `interface = Ftdi().open_from_url('ftdi:///1')`
with
`ftdi = Ftdi()`
`interface = ftdi.open_from_url('ftdi:///1')`

> Question - what is cause of the error "NameError: name 'ftdiEeprom' is not defined" when I try to inspect data in EEPROM?
I am following documentation section "eeprom - EEPROM API Quickstart".

You are not importing the module needed. Add the line `from pyftdi.eeprom import FtdiEeprom` after `from pyftdi.ftdi import Ftdi`

eabase, 2024-11-19:
@wb0gaz 
Please enclose your input, output and files in tripple back quotes, like this:
\`\`\`python
...
\<your code or output\>
...
\`\`\`

eblot, 2024-11-19:
> @wb0gaz Please enclose your input, output and files in tripple back quotes

I've edited the original post, hope this improves readability. Thanks @wb0gaz

eabase, 2024-11-20:
@eblot 
BTW. In Python you can print repeated characters with multipliers like this: `print('-'*40)`.

wb0gaz, 2024-11-20:
Thank you for posting additional detail - 

The problems I encountered at first with pyftdi were purely due to my relative inexperience with Python, particularly use in object oriented fashion (I am basically embedded C/Assembly programmer). I was also not aware that text of posting to github page would be interpreted with Python language elements for mark-up, so again I appreciate the advice/guidance. I may return for advice on specifics of pyftdi, but for now I am able to use pyftdi at beginner level and am not currently experiencing any problem.

wb0gaz, 2024-11-20:
As this was never a problem with pyftdi, only my inexperience with Python, I am closing this issue. If closing the issue is not correct step now, please let me know or adjust this posting as you see fit. Thank you again for the excellent support!

closed 2024-11-20 by wb0gaz