eblot/pyftdi #405: I2C Fake tri-state generate a glitch which can be considered as a Start

renaudwalder, 2025-03-18:
Using the fake tri-state of the device on an I2c transfer generates a glitch in the SCL and SDA signal.
After some investigation, it comes from the __clk_input_data_input_ commands.

Replacing this command by __idle_  command, solves the issue.

Here is some pictures of the stop bits and the glitch.

[image]

[image]

dnschneid, 2025-05-02:
I've encountered this bug as well.  The workaround we went with was to disable fake tristate mode (`I2cController._fake_tristate = False`), but obviously that's not ideal either.

Thanks for filing this bug; it sure saved us a lot of head-scratching.

dnschneid, 2025-05-07:
Looking at this again, it seems to me that `_idle` implements an active drive-high state, so that wouldn't be an appropriate substitute for `_clk_input_data_input` if you want fake tristate to work.  If you're happy with the behavior with _idle, you should just disable fake_tristate mode.

Changing `_clk_input_data_input` to have the data bits high (rather than low) eliminates the glitch and still presents high-impedance.  Technically, this replaces the low glitch with a high glitch, but as this is used after the stop condition, the bus is expected to go high regardless.

Similiarly, I can confirm that `_clk_lo_data_input` does generate a negative glitch on SDA (if it was high) before going high-impedance.  I think this behavior is preferable over the alternative (glitching high before going high-impedance), since the device might already be driving low on the bus to provide an ACK and we don't want brief crowbar currents.

Filed https://github.com/eblot/pyftdi/pull/414 with the fix.

PR #414 (2025-08-14): i2c: glitch high rather than low when tristating SCL+SDA
Avoids accidential start conditions, fixes #405
closed 2025-08-14 by eblot