eblot/pyftdi #409: How can I find "Tools"

lc2047, 2025-04-15:
Hi,

I am new to programming and eager to learn. I hope you can share your insights with me.

I came across the "Tools" section on [this page](https://eblot.github.io/pyftdi/tools.html) and would like to explore and study the example code, such as `ftdi_urls.py`. However, I am unsure where to find this file.

Could you please guide me or share your comments?

Thank you!

lc2047, 2025-04-15:
I get it.https://github.com/eblot/pyftdi/tree/master/pyftdi/bin

eblot, 2025-04-15:
Or if you install it with`pip`, in your environment bin directory:
````sh
> python3 -m venv venv
> . venv/bin/activate
(venv) > pip3 install pyftdi
Collecting pyftdi
  Downloading pyftdi-0.56.0-py3-none-any.whl.metadata (3.2 kB)
Collecting pyusb!=1.2.0,>=1.0.0 (from pyftdi)
  Downloading pyusb-1.3.1-py3-none-any.whl.metadata (2.5 kB)
Collecting pyserial>=3.0 (from pyftdi)
  Using cached pyserial-3.5-py2.py3-none-any.whl.metadata (1.6 kB)
Downloading pyftdi-0.56.0-py3-none-any.whl (145 kB)
Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)
Downloading pyusb-1.3.1-py3-none-any.whl (58 kB)
Installing collected packages: pyserial, pyusb, pyftdi
Successfully installed pyftdi-0.56.0 pyserial-3.5 pyusb-1.3.1
(venv) > l venv/bin/
__pycache__/        activate.fish       ftdi_urls.py*       pip3*               pyserial-ports*     python3@
activate            Activate.ps1        i2cscan.py*         pip3.13*            pyterm.py*          python3.13@
activate.csh        ftconf.py*          pip*                pyserial-miniterm*  python@
````

lc2047, 2025-04-16:
Hi eblot

thanks for your support. 
The pyFTDI was installed via pyCharm and I think the code is stored in some folder like "example" or "Tools" before. 
I found them in "\pyftdi-main\pyftdi\bin" later.

Hope you can help me.
I would like to use ft232h cjmcu to MX25V8035F and try to read memory chip information to confirm the SPI interface. But it always return 0xFF, that seem can't link with the chip.

Do you have SPI success example code for me?

BR

sifive-eblot, 2025-04-16:
Check out examples from the companion project `pyspiflash` available on GitHub and PyPI.

lc2047, 2025-04-16:
Hi eblot,

https://github.com/eblot/pyspiflash/tree/master
Thanks for your support. I will study it.

BR

lc2047, 2025-04-16:
Hi eblot

there have two serialflash.py one in /spiflash/serialflash.py and another one in /spiflash/tests/serialflash.py.

What is different between them?

eblot, 2025-04-16:
You need to post questions about pyspiflash in the [pyspiflash](https://github.com/eblot/pyspiflash) project, otherwise it will become hard to follow for other readers :-)