eblot/pyftdi #426: show_devices returns four devices with the same URL for FT4232H

edcloudcycle, 2025-09-02:
Hi,

I am working under Windows 11 and have used Zadig to assign the libusbK v3.1 driver to each of the channels I could see for the FT4232H I am using. These no longer appear as com ports as expected.

I then use `Ftdi.show_devices()` to get the device list and see:

```
Available interfaces:
  ftdi://ftdi:4232:FT7ZAX1O/1  (FT4232H-56Q MiniModule)
  ftdi://ftdi:4232:FT7ZAX1O/1  (FT4232H-56Q MiniModule)
  ftdi://ftdi:4232:FT7ZAX1O/1  (FT4232H-56Q MiniModule)
  ftdi://ftdi:4232:FT7ZAX1O/1  (FT4232H-56Q MiniModule)
```

I believe these devices should be unique in their serial numbers e.g. FT7ZAX1O**A**, so that I can identify them uniquely.

I am able to talk to each device but only by using the USB bus and address method:
```
i2c.configure('ftdi://ftdi:4232:00:FF/1')
```
If I attempt to use the URL returned by show_devices:
```
dev = Ftdi()
dev.open_from_url(url='ftdi://ftdi:4232:FT7ZAX1O/1')
```
I get an error because the name is not unique.
```
Exception has occurred: UsbToolsError
4 USB devices match URL 'ftdi://ftdi:4232:FT7ZAX1O/1'
```

I saw with the FT4222 library the serial numbers were returned with the channel character as expected.

All suggestions appreciated, ideally I would be able to identify the devices by channel.

Thanks

Ed

edcloudcycle, 2025-09-02:
Thanks @eblot that was really helpful, I had not seen pyftdiwin.

Once I got my drivers back to their default state I can see:
```
Available interfaces:
  ftdi://ftdi:4232:FT7ZAX1O/1   (FT4232H-56Q MiniModule)
  ftdi://ftdi:4232:FT7ZAX1O/2   (FT4232H-56Q MiniModule)
  ftdi://ftdi:4232:FT7ZAX1O/3   (FT4232H-56Q MiniModule)
  ftdi://ftdi:4232:FT7ZAX1O/4   (FT4232H-56Q MiniModule)
```

Which is much better. Unfortunately now one of the GPIO commands that worked before does not work. I will investigate further.
```
i2c = I2cController()
i2c.configure('ftdi://ftdi:4232:FT7ZAX1O/1')
slave = i2c.get_port(0x21)
i2c.set_gpio_direction(0xf8, 0xf8)  # set upper 5 bits as output
i2c.write_gpio(0xf8)

Exception has occurred: I2cIOError
Cannot read GPIO
```

eblot, 2025-09-03:
It really depends. 
If there are fixes, patches are always welcome. 
If they break existing APIs, it is likely they need to be reworked first... 
I can't really tell without knowing their content first.
What are they about? Thanks.

eblot, 2025-09-03:
Have you had a look at [Zadig](https://eblot.github.io/pyftdi/installation.html#id1) caveat? It looks like something similar to your issue, but to be honest I tend to stay as far as possible from Windows that I can.

This is also why I do not want to add support for D2XX: I have no way to test it, and I really do want to spend time with M$. I've been able to stay away from it since 2003 and I really hope to keep it this way :-)

I understand this can be frustrating for Windows users. However till someone steps up and becomes a trustworthy pyftdi maintainer for Windows support, I'm afraid pyftdi may not be able to properly support it.

closed 2025-09-03 by edcloudcycle